open-sdr/openwifi #98: SDR frequency

c2013hl, 2021-09-10:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?
From Lee - c2013hl@gmail.com: Hi, I am looking to do some experiments with SDR using openwifi on Analog Device ADRV9361-Z7025 hardware. Mainly to understand the flexibility of the hardware with openwifi.

1. Our image is used directly or you build your own image?
will need to build own image with slight modification. currently trying to experiment with TX and RX frequency variations

2. What is your own modification?
Would you give some advice how I can change openwifi to different frequency band. in driver file: hw_def.h, it listed 900MHz to 60GHz. How can I change the band for openwifi to 900MHz or 2.4GHz? It's currently default to 5GHz.

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
OS - ADI Linux; Vivado 2018.3 same as openwifi, openwifi from main branch

4. Board/hardware type
ADRV9361-Z7035 from ADI

5. WiFi channel number
tx 1a, rx 1a

6. Steps to reproduce the issue, and the related error message, screenshot, etc
No issue. The change is intended for experimentation

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
Not applicable.

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods
Read hardware and software code on openwifi

9. Any other thing we need to know for helping you better? 
no.

JiaoXianjun, 2021-09-12:
Hello,

In the https://github.com/open-sdr/openwifi#quick-start , we use https://github.com/open-sdr/openwifi/blob/master/user_space/hostapd-openwifi.conf or https://github.com/open-sdr/openwifi/blob/master/user_space/hostapd-openwifi-11ag.conf to configure the AP's channel. But currently we only test in 5GHz. Not sure in 2.4GHz how good/bad it is.

Regarding changing the actual working frequency to the non-standard WiFi channels, we never try this. But I think it is doable by yourself, after you understand the whole frequency setting mechanism as we explained: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#regulation-and-channel-config

c2013hl, 2021-09-13:
thank you for your reply. i will look at the info you suggested. if tested
successfully, I will contribute an application note.

JiaoXianjun, 2021-09-13:
Good. Looking forward to your result.

c2013hl, 2021-09-22:
We tried to change the frequency on the 2 SDR boards (ADRV9361-Z7035) but
no luck.
echo, sync, cat does show the correct frequency setting.

However, The default frequency is 5.2GHz for openwifi.
We changed the TX/RX LO of both the Access Point and Client to 2.4GHz then
apply the instructions from ap-client-two-sdr.md apnote.
When the client node tries to "iwlist sdr0 scan" or run "wpa_supplicant
-i sdr0 -c wpa-openwifi.conf", the client only uses 5.2GHz frequency and
does not use 2.4GHz.
We did try to change the .conf file but no luck.

Would you give us some advice. Thank you.

JiaoXianjun, 2021-09-30:
If you could share us the dmesg log, we can take a look.

c2013hl, 2021-10-01:
Thank you for the response. We are able to set frequency to 2.4GHz now.
5.2G and 2.4G seems to work. We're trying some other frequencies. It looks
like we need to change the device driver to be able to support other
frequencies.

c2013hl, 2021-10-07:
We tried some other frequencies different than the standard 5G and 2.4G
(for example: 3GHz or 900MHz)
would you advise what we need to change in the driver to support this kind
of setting and scanning?
Also which part of the driver supports 'iwlist scan" function? Would you
help to point us to the code files that may be responsible for doing the
scan?

JiaoXianjun, 2021-10-08:
Hello,

All frequency setting related HW SW blocks are explained here: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Analog-and-digital-frequency-design

You need to figure out and try to change related files.

After you run "iwlist scan", you can run "dmesg" to see those printing messages (you will see openwifi is switching channels one by one). Then search those printing messages in the sdr.c, you will find out which functions are called to do the frequency scanning.

c2013hl, 2021-11-02:
> If no further question, I will close this issue.

Ok, thanks.

closed 2021-11-04 by JiaoXianjun

t109368035, 2021-12-07:
Hello @c2013hl ,
How did you set the frequency(Access Point) to 2.4GHz?
thank you!

t109368035, 2021-12-07:
Hello  @c2013hl ,
Which driver is used for frequency 2.4G?
Thanks!