open-sdr/openwifi #99: Possibility about AD936X module for PYNQ-Z1-like low-cost boards

regymm, 2021-09-12:
Hi, I'm a student interested in FPGA and want to learn more about SDR, especially AD936X based ones. I wonder if it's possible to have AD936X extension boards for low-cost FPGA platforms like the PYNQ-Z1, for example, using PMOD or the Arduino-like connectors -- as it's the only big enough ZYNQ I have, and cheaper than most available SDRs anyway. 

In the PlutoSDR schematics, there seem to be 42 IOs between 9363 and 7020, and part of them are not high-speed ones. Meanwhile, there are 49 GPIOs available on PYNQ-Z1 and similar Arty-Z7: they are scattered here and there but enough. Frequency can be a big problem -- I didn't find how high the frequency of 9363 IOs are so far(seems the datasheet is only an introduction), and PCB traces of these GPIOs are not length-matched. I'd appreciate it if help about this part can be provided. 

Also, power of PYNQ-Z1 IOs are 3V3, but the power supply chip is programmable and I think it's possible to re-flash it to output 1V8 for Bank34/Bank35. The IOs also have series resistors but can be easily removed/shorted. 

The final product will certainly be ugly because of GPIO form factors, but if frequency permits I think it's worth having a try(using scavenged cheap AD9363 and JLC's 4-layer PCB).

JiaoXianjun, 2021-09-12:
Hello,

Thanks for your interest in openwifi.

Unfortunately, we are not hardware designer, and have less experiences on hardware. So, hard to answer your question. Maybe using CMOS mode instead of LVDS mode could be considered? Because CMOS mode needs lower speed. (Of course also limit the sampling rate).

Regarding the 7020+9363 hardware, as far as we know, the hardware mentioned in https://github.com/open-sdr/openwifi/issues/91 is the cheapest. It might be the same price range as your target pynq-z1 + ad9363?

regymm, 2021-09-12:
Thanks for your reply. I'll have a look at the CMOS mode. 

I'm afraid AntSDR is still more expensive than PYNQ-Z1 + AD9363(if possible) -- PYNQ-Z1 at 900 RMB and a scavenged, not-yet-reballed 9363 may cost as low as 30 RMB per piece.

JiaoXianjun, 2021-09-12:
~1000RMB? Then it will make a lot of sense, considering many people already have pynq-z1.

Besides the LVDS/CMOS interface to AD9361 ADC/DAC, we also need AD9361 CTRL GPIO connected to FPGA. If all these pins work with FPGA, then openwifi should run (of course after changing the pin map/constrains).

You can check how other boards connect ADC/DAC LVDS/CMOS and CTRL GPIO to FPGA. I have an impression that adrv9364 7020 SOM also support CMOS mode: https://github.com/analogdevicesinc/hdl/tree/master/projects/adrv9364z7020

Maybe that could be a good reference/starting-point for you.

regymm, 2021-09-13:
Yes, I'll begin with existing designs. Though it may take a long time before I start working on my own board, I'll keep this project in mind.

summershrimp, 2021-09-26:
PYNQ-Z1 is not possible to use openwifi, at lease use  PYNQ-Z2 which soc is zynq7020

JiaoXianjun, 2021-09-30:
> PYNQ-Z1 is not possible to use openwifi, at lease use PYNQ-Z2 which soc is zynq7020

According to https://www.xilinx.com/products/boards-and-kits/1-hydd4z.html , Z1 has 7020

regymm, 2021-10-11:
After thinking for a few days, I think this is not possible without destructive and cost-inefficient modifications. There are enough diffpairs if using 2 PMOD and 2 HDMI, and other signals can be safely routed via long wires. However, voltage is the problem -- the 7020 bank voltage seems to have to be set to 2.5V(from 3.3V) for AD9363 LVDS interfacing, and with 3.3V missing, I guess both gigabyte ethernet and USB Phy, probably FT2232H JTAG and UART as well, will fail if not be damaged(eg. USB Phy has its own 3V3 LDO output, so will collide with 2V5). And having another USB Phy using GPIOs left by AD9363 and accessed via EMIO just doesn't feel sane. 

Anyway, I've bought some recycled AD9363, and will stash them for future use instead of attempting an extension for PYNQ-Z1.

regymm, 2021-11-12:
Just a status update, the extension board has arrived and initial testing show that AD9363 can be recognized in ported plutosdr Linux. So the level shifters are working promisingly. Major challenge(aka LVDS translation) haven't been tested yet, but I'll keep on going. 

[image]

JiaoXianjun, 2021-11-12:
Exciting!

JiaoXianjun, 2022-04-01:
Any news about your SDR HAT commercial availability?

regymm, 2022-04-06:
Hi, I submitted another batch of five to PCB/SMT last month, but due to the covid19 situation I haven't received them yet.

Flynnsin, 2022-06-20:
我使用antsdr板子不知道用哪个内核配置文件

base0010, 2022-08-17:
Hey @regymm any updates on this? Very interested.

regymm, 2022-08-18:
Hi, thanks for your interest! I'm still working on this, though progress in recent months have been slow. I'm now debugging existing boards(so far so good) and manufacturing small bridge boards(because old ones have been used up). You can follow updates on my Twitter.