open-sdr/openwifi #100: Question for CSI document, what is 'equalizer'

Eugene-Liu-milkshake, 2021-09-13:
From Eugene Liu:

     Hi, I am looking to do some experiments with SDR using openwifi on Analog Device adrv9364z7020 hardware. I am first grade master student  and new to this device. Mainly to understand presence detection by using channel state information(CSI). 
     I had read the CSI document on openwifi github, and curious about what is the 'equalizer' meaning. I had spend lots of time survey on internet but still not clearly. At first, I think the constellation is represent the value which is used to equalizer raw data, but not sure about it.
     Could you give me some knowledge about equalizer?

    Thanks for your openwifi project, this quite helpful for us to improve our research.

    Sincerely,
    Eugene Liu.

JiaoXianjun, 2021-09-13:
Equalizer is the signal processing step that does equalization on the channel distorted received sub-carrier signal.

For instance:

The received sub-carrier signal is r, and the channel gain on that sub-carrier is h, then the equalizer (zero-forcing in our case) does:
```
r/h
```
to remove the channel effect and recover the symbol (qpsk, 16qam, 64qam) transmitted on that sub-carrier at the the transmitter side.

JiaoXianjun, 2021-09-18:
If no further question, I will close this issue.

Eugene-Liu-milkshake, 2021-09-18:
From Eugene Liu:

     Thanks for your reply, now I have clearly understood about what is equalizer. Now I have displayed csi and equalizer figures successfully by your program.
     Equalizer equal to r/h. From my understanding, equalizer is inversely proportional to channel gain. According to this relationship, I can understood the result is bad when equalizer value is too big. But I am not sure why need to set equalizer threshold value as 2000.
I am not sure how to define "good equalizer". 

at side_info_display.py at line 39
# only take out the good equalizer result, when output > 2000, it is not good

Thanks again for your time.

Sincerely,
Eugene Liu.

JiaoXianjun, 2021-09-23:
2000 is a limit for "good looking" display according to the internal normalization of FPGA code. If I remember correctly, the equalization output constellation points are normalized in a 1024*1024 square (or 2048*2048? don't remember precisely)

When doing r/h, actually we don't know the actual h except that we estimate it based on LTF.
So there will be estimation error on h. Sometimes, with the estimation error, the estimated h could be very small (close to 0), especially when that subcarrier is in deep fading. In this case, the r/h will be super big (divided by ~0), so it will be a occasional point which is very far away from the normal constellation square. So currently this occasional bad point is dropped to let us focus on the normal constellation points/drawing.

Eugene-Liu-milkshake, 2021-09-24:
I got it, thanks for your answer = U =

closed 2021-09-24 by JiaoXianjun