open-sdr/openwifi #101: Using opensource EDA

ne-vlezay80, 2021-09-24:
Hello, why don't you use instead of the commercial and closed Vivado, its open counterparts, such as gEDA, icarus verilog, kiCAD ...

The fact is that the cost of a license for Vivado is very expensive and for some people it is simply unaffordable. Also, the use of closed source software on open systems looks inappropriate and also violates ethical standards.

The cost of a Vivado license is $1995, which is about half an apartment in Abkhazia.

JiaoXianjun, 2021-09-24:
If you use those platforms that have 7020 Zynq FPGA (antsdr, adrv9364z7020, zedboard, zc702) listed on the table in our README, then vivado is for free (not any paid license is needed).

There are lots of things on our todo list. Opensource EDA tool definitely is on it. But it is not at the 1st location. Currently we focus on making the openwifi as good/mature as COTS wifi chip.

Make it work well firstly, then we will consider other things.

But of course, if someone want to try openwifi with opensource EDA, that will help us a lot!

Localacct21, 2021-10-11:
Which free version of vivado should we use then? The web veiw? OR the 30day preview?

JiaoXianjun, 2021-11-30:
If no further questions, I will close this issue.

regymm, 2021-12-01:
(Since I've tried open-source FPGA flow recently, I'll say what I know, which may be inaccurate)
Currently, the SymbiFlow project supports a full open-source dev flow on some FPGAs, xc7a200t is the largest one. About ZYNQ, 7010 and 7020 are supported but I don't know how well PS/PL interaction is supported. [Seems EMIO and some AXI lite are working](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/xc/xc7/tests/ps7). I guess it's still far away from Openwifi's requirements. 

If wanting a completely free toolchain, *maybe* one approach is to put an open-source RISC-V softcore acting as ZYNQ in xc7a200t? Or, like traditional ways, using an Artix 7 and another standalone processor(with open-source toolchain, I don't know if this kind exist). I think these two both require radical changes to the current code. 

Or we could just wait until SymbiFlow's conquering of xc7z020 PS/PL interfaces.

blacklightpy, 2023-09-03:
> Or we could just wait until SymbiFlow's conquering of xc7z020 PS/PL interfaces.

@regymm SymbiFlow is now F4PGA

ne-vlezay80, 2024-05-08:
Is it passible openroad[https://github.com/The-OpenROAD-Project] support

blacklightpy, 2024-05-09:
> Is it passible openroad[https://github.com/The-OpenROAD-Project] support

Isn't that for ASICs?
Apparently it supports various PDKs including Google's SkyWater 130nm and has been tested on proprietary PDKs upto 12nm. It also supports ASAP7 7nm finFET PDK.