open-sdr/openwifi #103: AP is ENABLED but not shown on other device scanning list using ADRV9361Z7035 even taking very close to Tx antennas why?

waseem-arshad94, 2021-10-08:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

waseem-arshad94, 2021-10-13:
0. Yes I will send you the Introduction e-mail.
1. I used your image directly.
2. I will modify the design later when I will run it on my board.
3. Ubantu 18.04, Vivado 2019.1,2019_R1-2020_02_04.img.xz.
4. ADRV9361-Z7035 + ADRV1CRR-FMC I am using.
5. WIFI channel I am using is Tx1Rx1.
6. Device SSID "openwifi" is not showing in the scanning of other device or laptop.
7. I did not understand this one.
8. I read carefully and apply your method in README file multiple time but id did not get SSID.
9. You need to help me about this I want to just receive the wifi bitstream not be transmit side. Please help me how I can receive the wifi signal on my board. After receiving the bitstream I will extract the MAC information from received bitstream. Thanks in advance for helping me.

waseem-arshad94, 2021-10-18:
Dear Sir;
I tested according your above suggestions :results of this: cat /proc/interrupts (Execute the "cat ..." command for several times)
(You should see the number of "sdr,tx_itrpt1" grows, because it sends the "openwifi" beacon periodically)
- It produce the error -bash: cat   command not found.
- secondly AP is not enabling which should be AP-ENABLED after the running ./fodsem.sh file.
- Both boards on AP side and Client side are working and pinging with their PCs but not shown "sdr0" as interface.
- hostapd can't started error is generation at the end.
Thanks

waseem-arshad94, 2021-10-18:
Dear sir;
-I resolved the AP-ENABLED and "sdr0" interface issues which are mentioned above my post.
- Now on client side board after running "iwlist sdr0 scan"  generating error device busy or interface can't support scanning.
- Thanks

JiaoXianjun, 2021-10-18:
Would you please paste all the commands/steps you run at the client side? Only having these info, we can help.

waseem-arshad94, 2021-10-20:
Dear Sir;
I run following commands at client side:
ssh root@192.168.10.122
(password: openwifi)
service network-manager stop
cd openwifi
./wgd.sh
(Wait for the script completed)
ifconfig sdr0 up
iwlist sdr0 scan
(The "openwifi" AP should be listed in the scanning results)
wpa_supplicant -i sdr0 -c wpa-openwifi.conf
("iwconfig sdr0 essid openwifi" could also work. Less info compared to wpa_supplicant)

waseem-arshad94, 2021-11-01:
Dear Sir;
I have few analysis about openwifi. It scan one channel at a time. Can we scan multiple channel with sdr0?

**Important Note**
In normal wifi hacking in linux operating system we use **airmon** utility to setup in monitor mode like: **airpdump-ng** for wifi scanning. Can we use this utility for wifi scanning at sdr0 in monitor mode.

Thanks in advance!

JiaoXianjun, 2021-11-01:
> Dear Sir; Thanks for your kind help! Now my both sdr are in working condition. Now I have few questions in my mind kindly clear my points.
> 
> 1. In scan list only my device is scan with (SSID="openwifi", freq=5220Mhz) not other wifi routers in my building.
> 2. How we can shift my sdr frequency from 5 GHz to 2.4 GHz?
> 3. How my sdr will scan all wifi present in building?

To scan all bands/frequencies, you need connect dual-band (2.4/5GHz) antennas to your SDR tx and rx.

JiaoXianjun, 2021-11-01:
> Dear Sir; I have few analysis about openwifi. It scan one channel at a time. Can we scan multiple channel with sdr0?
>
> **Important Note** In normal wifi hacking in linux operating system we use [...]

iwlist does scan wifi channel one-by-one.
openwifi does support monitor mode, and it behaves like other COTS WiFi card. So you can use any existing Linux tools (install them or apt install) on top of openwifi card.

We also have monitor and packet injection introduction in the app notes: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/README.md  (802.11 packet injection and fuzzing)

waseem-arshad94, 2021-11-03:
Dear sir;
I need to open the design in VIVADO How it can be done easily?
Thanks in advance!

waseem-arshad94, 2021-11-04:
Dear sir;
I want to build the HDL Projects in VIVADO for customization the logic of openwifi. How I can run the block design of openwifi in VIVADO and which version I needed to build it.

JiaoXianjun, 2021-11-04:
https://github.com/open-sdr/openwifi-hw

waseem-arshad94, 2021-11-05:
Dear Sir;
During Building I found **ERROR**
Building adrv9361z7035_ccbob_lvds project [/home/ijaz/openwifi_hw/openwifi-hw/adi-hdl/projects/adrv9361z7035/ccbob_lvds/adrv9361z7035_ccbob_lvds_vivado.log] ... FAILED
For details see /home/ijaz/openwifi_hw/openwifi-hw/adi-hdl/projects/adrv9361z7035/ccbob_lvds/adrv9361z7035_ccbob_lvds_vivado.log

I Opened this **adrv9361z7035/ccbob_lvds/adrv9361z7035_ccbob_lvds_vivado.log** and found at the end of file:
ERROR: [Common 17-345] A valid license was not found for feature 'Synthesis' and/or device 'xc7z035i'. Please run the Vivado License Manager for assistance in determining
which features and devices are licensed for your system.
Resolution: Check the status of your licenses in the Vivado License Manager. For debug help search Xilinx Support for "Licensing FAQ". If you are using a license server, verify that the license server is u$
INFO: [Common 17-206] Exiting Vivado at Fri Nov  5 12:21:25 2021...
[Fri Nov  5 12:21:26 2021] synth_1 finished
wait_on_run: Time (s): cpu = 00:00:08 ; elapsed = 00:00:08 . Memory (MB): peak = 2095.457 ; gain = 0.000 ; free physical = 24954 ; free virtual = 30019
ERROR: [Common 17-69] Command failed: Run 'synth_1' failed. Unable to open
INFO: [Common 17-206] Exiting Vivado at Fri Nov  5 12:21:26 2021...

JiaoXianjun, 2021-11-05:
As the error said, you need valid vivado license. Vivado is only free for small fpga, such as zedboard, zc702, antsdr, adrv9364z7020. Your 7035 fpga needs vivado license.

waseem-arshad94, 2021-11-05:
@JiaoXianjun Hi,
How I can get it easily?

JiaoXianjun, 2021-11-05:
You need to buy the license from Xilinx or via your company/university.

waseem-arshad94, 2021-11-05:
Dear Sir;
I am using the licensed version of vivado 2018.3. Why I am getting the this error. My license is valid up to 2023. We purchased it from Xilinx.

JiaoXianjun, 2021-11-05:
If that is the case, I would suggest you ask Xilinx.

waseem-arshad94, 2021-11-08:
Dear sir;
I have few **questions more** about openwifi:

1. How we can access the IOs of board in the openwifi?
2. Can we run the Qt based GUI for openwifi if yes then how we can?
3. You said previously we can scan all bands 2.4/5.8 GHz using dual band antenna. Is default image is capable of scanning all bands after just putting the antenna or any other scripts or instruction needed to tell the board to scan all bands at time?
4. Is any test example to check openwifi in monitor mode?

waseem-arshad94, 2021-11-09:
Dear Sir;
I opened the design in Vivado 2018.3 successfully. After that I have a question in my mind:

1. wifi stack implementation is in PL or PS because I didn't see any **modulation or demodulation in FPGA design**?. 
2. How your FPGA design communicate with PS?
3. Is this fpga design default of ADI reference design or any other modification you done in this FPGA design?
4. I didn't see any .c files in my project files.

JiaoXianjun, 2021-11-09:
Hi,

Maybe it would be good for you to read (word by word):
https://github.com/open-sdr/openwifi/blob/master/README.md
https://github.com/open-sdr/openwifi-hw/blob/master/README.md
https://www.orca-project.eu/wp-content/uploads/sites/4/2020/03/openwifi-vtc-antwerp-PID1249076.pdf
https://users.ugent.be/~xjiao/Cedric_Den_Haese_masterproef.pdf
https://arxiv.org/abs/2109.03032
https://ieeexplore.ieee.org/document/9573364

And watch videos:
https://youtu.be/Mq48cGthk7M
https://media.libreplanet.org/u/libreplanet/m/openwifi-project-the-dawn-of-the-free-libre-wifi-chip/

All your questions can be answered I believe by reading/watching above materials.

I will close this issue, because the topic has been changed to learning. Basically you can't master everything in 1 day. Be patient and put effort/time, I believe you will have answers of all your questions.

After all the above reading/watching, if you still have questions, you can open an new issue.

Thanks.

closed 2021-11-09 by JiaoXianjun