open-sdr/openwifi #107: Configuration for 802.11n AP

Jason414, 2021-10-18:
I am using Zedboard running with openwifi on it. I am willing to run it as 802.11n (or 802.11g) AP with 2.4GHz. I have changed file hostapd-openwifi.conf as following 

interface=sdr0
driver=nl80211
country_code=US
ssid=openwifi
hw_mode=n
channel=6
supported_rates=60 90 120 180 240 360 480 540
basic_rates=60 90 120 180
ieee80211n=1
ht_capab=[SHORT-GI-20]
require_ht=1
#ieee80211d=1
#ieee80211h=1
#wpa=2
#wpa_passphrase=myrabbit
#wpa_key_mgmt=WPA-PSK

But after I run command ./fosdem.sh, it says 
Configuration file: hostapd-openwifi.conf
Line 5: unknown hw_mode 'n'

Could anyone help to correct the configuration file to access 2.4GHz frequency?

JiaoXianjun, 2021-11-08:
Have you tried it successfully?

JiaoXianjun, 2021-11-30:
If no further questions, I will close this issue.

closed 2022-04-01 by JiaoXianjun