open-sdr/openwifi #108: post_config.sh failes during the adrv9364z7020 configuration

MamonthPenis, 2021-10-18:
I built the Linux img from scratch, burned it on SD card and finally when I am running post_config.sh I have this:

root@analog:~# cd ~/openwifi && ./post_config.sh
++ uname -m
+ MACHINE_TYPE=armv7l
++ uname -r
+ '[' -d /lib/modules/4.14.0-gb6e379910a11-dirty ']'
+ '[' armv7l == aarch64 ']'
++ uname -r
+ ln -s /lib/modules/adi-linux /lib/modules/4.14.0-gb6e379910a11-dirty
+ depmod
depmod: WARNING: could not open modules.order at /lib/modules/4.14.0-gb6e379910a11-dirty: No such file or directory
depmod: WARNING: could not open modules.builtin at /lib/modules/4.14.0-gb6e379910a11-dirty: No such file or directory
+ modprobe mac80211
modprobe: ERROR: ../libkmod/libkmod.c:514 lookup_builtin_file() could not open builtin file '/lib/modules/4.14.0-gb6e379910a11-dirty/modules.builtin.bin'
modprobe: FATAL: Module mac80211 not found in directory /lib/modules/4.14.0-gb6e379910a11-dirty
root@analog:~/openwifi# root@analog:~# cd ~/openwifi && ./post_config.sh
-bash: root@analog:~#: command not found
RNING: could not open modules.builtin at /lib/modules/4.14.0-gb6e379910a11-dirty: No such file or directory
+ modprobe mac80211
modprobe: ERROR: ../libkmod/libkmod.c:514 lookup_builtin_file() could not open builtin file '/lib/modules/4.14.0-gb6e379910a11-dirty/modules.builtin.bin'
modprobe: FATAL: Module mac80211 not found in directory /lib/modules/4.14.0-gb6e379910a11-dirty

'/lib/modules/4.14.0-gb6e379910a11-dirty/modules.builtin.bin  is present, but it's size is 0 bytes. The same problem is with other files which "could not be opened".

I am using adrv9364z7020 board with the last 2021-07-28-ADI-Kuiper-full.img on sd card. Xilinx sdk 2018.3.

JiaoXianjun, 2021-10-19:
After running this script (live with error), can you reboot and run openwifi successfully?

JiaoXianjun, 2021-10-19:
Hello,
I think we need to take a look at all the steps and outputs when you build openwifi image based on: 2019_R1-2020_02_04.img.xz . Could you share them here?

JiaoXianjun, 2021-10-19:
Hello,

Seems that you have some old stuffs in place, and they prevent adopting the newly generated files.

Such as:

```
Kernel found! Skip the time costly Linux kernel compiling.
...
++ sudo mkdir /media/pentest/rootfs/root/openwifi/drv64
mkdir: cannot create directory ‘/media/pentest/rootfs/root/openwifi/drv64’: File exists
```

You can do following operations to make sure you have a fresh new openwifi image:
0. Restore 2019_R1-2020_02_04.img.xz to your SD card
1. Clean the adi-linux directory:
```
cd openwifi/adi-linux
git reset --hard
git clean -dxf
cd openwifi/adi-linux-64
git reset --hard
git clean -dxf
```
2. Then follow the "Build openwifi Linux img from scratch"

MamonthPenis, 2021-10-19:
Yes, I have described what happened after I followed your suggestions.