open-sdr/openwifi #111: Can npt establish connection to AP

MamonthPenis, 2021-10-21:
Using adrv9364z7020 board. Trying to connect to Wifi network:

route del default gw 192.168.10.1
root@analog:~/openwifi# wpa_supplicant -i sdr0 -c wpa-connect.conf &
[1] 2510
root@analog:~/openwifi# Successfully initialized wpa_supplicant
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SCAN-STARTED
random: Cannot read from /dev/random: Resource temporarily unavailable
handle_probe_req: send failed
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MyVW" auth_failures=1 duration=10
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SSID-REENABLED id=0 ssid="MyVW"
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MyVW" auth_failures=2 duration=20
sdr0: CTRL-EVENT-SCAN-STARTED
handle_probe_req: send failed
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SSID-REENABLED id=0 ssid="MyVW"
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MyVW" auth_failures=3 duration=20
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SSID-REENABLED id=0 ssid="MyVW"
sdr0: SME: Trying to authenticate with 1a:48:ca:b0:02:93 (SSID='MyVW' freq=2472 MHz)
sdr0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MyVW" auth_failures=4 duration=20
sdr0: CTRL-EVENT-SCAN-STARTED

JiaoXianjun, 2021-11-01:
> Yes, creating an AP in 5G band is working. But still can not connect to network from openwifi board.

Can you describe more in detail about your 5GHz experiment?
What is the AP device?
What is the Client device?
What commands/operations did you run on AP and Client?
What is the error message?

JiaoXianjun, 2021-11-30:
If no further questions, I will close this issue.

closed 2022-04-02 by JiaoXianjun