open-sdr/openwifi #115: Quick start: SSH connection failed

gxhen, 2021-10-29:
Hi,

Firstly thanks for the open-Wi-Fi project. I encountered a problem during implementation.

I followed the procedure in 'quick start':
1. Download img file and burn it into the SD card. (32bit img for the combination of Xilinx zed board + FMCOMMS2/3/4)
2. Copy files from two folders to the base directory of BOOT partition. (Some files in base directory are replaced) 
3. Confirm Zedboard is in SD card boot mode.
4. Login to the board via SSH connection. (MobaXterm on a Windows PC)

The SSH connection cannot be established and it always reports an error of time out. 

**However, if I skip Step 2 and perform Step 1,3,4, the SSH connection was successfully established.** Although the board failed to find the device sdr0 in the following steps.

My board is Xilinx zed board + FMCOMMS2/3/4. Could you please guide me what is the possible reason? Many thanks.

JiaoXianjun, 2021-11-01:
"Copy files from two folders to the base directory of BOOT partition. (Some files in base directory are replaced)"

Can you tell me the exact name/path of the "two folders"?

gxhen, 2021-11-01:
Thanks for your quick reply. The two folders and file names are:
1. BOOT (E:)/openwifi/zed_fmcs2/  : BOOT.BIN and devicetree.dtb
2. BOOT (E:)/openwifi/zynq-common/ : uImage

I copied the above three files to BOOT (E:) and previous files were replaced. Then the SSH connection cannot be established anymore.

Many thanks for your time!

JiaoXianjun, 2021-11-01:
When you can not ssh the board, can you ping the board?
Can you share me the error message/log of ssh and ping?

gxhen, 2021-11-01:
Thanks very much. I did a quick test just now.  As shown in the attachments, I can ping the board successfully before coping those files, but failed after the copying operation. It shows 'Request timed out' on the cmd window.
[image]
[image]

JiaoXianjun, 2021-11-01:
Seems like you are using windows. Sometimes the file operation under windows has issue. Can you do the copy on board?

ssh to the board (do not copy files), then:
```
mount /dev/mmcblk0p1 /mnt
cd /mnt
```
Now you are in the BOOT partition. Copy those files (by Linux bash commands) to /mnt/
/mnt/openwifi/zed_fmcs2/ : BOOT.BIN and devicetree.dtb
/mnt/openwifi/zynq-common/ : uImage

Finally, run:
```
sync
cd /
umount /mnt
reboot now
```

Power cycle.
Try again ping and ssh.

gxhen, 2021-11-02:
Thanks a lot for your support. As shown in the picture, I have done the copying operations on the board based on your instructions.  But no improvement was made. After power cycling, It still reports ' request timed out ' when I try to pin the board. Should I try a PC with Linux system to burn the img file?
[image]

Many thanks for your time.

JiaoXianjun, 2021-11-02:
Yes. Better to burn the sd card in Linux (by the Disks program or dd) and do file copy in Linux. 

We never try windows.

gxhen, 2021-11-02:
Thanks for your reply. It seems due to the windows. I also tried to burn the SD card via 'sudo dd' command in OSX system but ran into the same problem. Will try Linux in the next few days and update here. 

Thanks again for your time.

gxhen, 2021-11-03:
Hi, I have tried on two Linux PCs, one is Ubuntu and another is Mint. Unfortunately, we got the same error. We can ping the board before copying those files but not after that.  I tried to do the copying both on PCs and the board but got the same result. It shows 'destination host unreachable' in terminal when I ping the board.

BTW, I also observed that the LED on the Zedboard labelled ''LD 9''/''MIO 7'' does not work after the copying operation (it works before copying). Could this reveal the reason?

Many thanks for your effort.

JiaoXianjun, 2021-11-04:
We have tried this procedure for many times. Could be some reasons at your side:

1. The downloading issue. Could you re-download the image file?
2. The SD card issue. Could you find another SD card?
2. Can you use gnome-disk-utility in Ubuntu to restore the image to your SD card? (Can you share us the exact dd command you used to restore the SD card?)

gxhen, 2021-11-04:
Thanks very much for your generous help. I have downloaded the image file several times thus this should not be the problem. I have just tried the disk-utility to burn the SD card but got no improvement. Previously I used the command 'sudo dd bs=4M if=<img path> of=/dev/sda' to restore the SD card.

I will try another SD card and reply to you. BTW, will the SD card reader and memory size affect this? I am using a USB card reader and a 32GB SD card.

JiaoXianjun, 2021-11-04:
Indeed, we never try 32GB SD card. We only tried 8G/16GB SD card before.
When you dd, could you use  bs=512? (This is the original parameter in the ADI original image)

gxhen, 2021-11-04:
No problem, I will try it.

JiaoXianjun, 2021-11-04:
I just downloaded the image and re-do everything at my side on zedboard+fmcomms2. Everything is OK.

gxhen, 2021-11-04:
Many thanks for your time. I will repeat all the steps and try to figure out the problem.

JiaoXianjun, 2021-11-08:
If you have tried it successfully, please tell us. Then we can close this issue.

gxhen, 2021-11-08:
Thanks for your ask. In fact, we did not address the problem successfully. We plan to try on another new Zedboard this week, and an antsdr in the next two weeks. It's okay to close this issue for now if you want, and I will open a new one when I get in trouble again. Many thanks for your support.

JiaoXianjun, 2021-11-08:
Take your time. 
I will leave this issue open.

gxhen, 2021-11-15:
Hi Dr Jiao, I have successfully solved the problem, which is due to the SD card adapter. I changed a new one and it works well now. Feel free to close this issue and thanks very much for your continued support.

JiaoXianjun, 2021-11-15:
Glad to hear the good news.
You are welcome!

closed 2021-11-15 by JiaoXianjun