open-sdr/openwifi #116: 传输功率修改

junziqiang, 2021-11-07:
焦老师，请问如果我想要修改传输功率除了在应用层使用iwconfig命令之外，我在源码这里应该怎么修改？以及能否根据不同的链路设置不同的传输功率？多谢焦老师

JiaoXianjun, 2021-11-09:
Hello,

1. In sdr.c, we use ad9361_set_tx_atten() to set AD9361 TX attenuation, which changes the TX power
2. You can also apply a digital gain factor via register 13 of tx_intf: https://github.com/open-sdr/openwifi-hw/blob/aa09fb8b23cb3873431f3c21e9faa5258b46e665/ip/tx_intf/src/tx_intf.v#L504 -- too small gain will result in big DAC quantization noise. Please search "tx I/Q digital gain before DUC" in this doc https://github.com/open-sdr/openwifi/blob/master/doc/README.md to see how to set this register via sdrctl.

If you mean "per link" by "per communication target MAC address", I think it is doable. You need to work further on driver (sdr.c) and FPGA.

JiaoXianjun, 2021-11-30:
If no further questions, I will close this issue.

JiaoXianjun, 2022-04-01:
You can modify Tx power (and many other things) easily by:
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#Tx-power-config