open-sdr/openwifi #117: Packet injection function cannot send data packets continuously

brealin, 2021-11-09:
I use the DEMO board of ADRV9361-Z7035, which has fully run the openwifi function, and my mobile phone can be connected to this board normally, and the throughput test can also reach 30Mbps. I want to monitor the transmit power and signal quality. I study the operation of this page to use the packet injection function to continuously send packets, https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md. I used a spectrum analyzer to monitor the output signal, but it did not work as expected. The signal on the spectrum analyzer only appears occasionally, and is not continuously sent according to the script setting. I visually observed that the sending frequency is 3-5S and a signal appears. I tried disabling ACK (sdrctl dev sdr0 set reg xpu 11 16 ), but there was no obvious change.

JiaoXianjun, 2021-11-09:
Hello,

1. When you run the injection program, you can specify "-n 999999" to have you "endless" packets
2. If you use another COTS WiFi device running monitor mode with wireshark, you should see lots of packets in the channel. This means openwifi is transmitting. Do please verify via another COTS WiFi monitor with wireshark.
3. Make sure your spectrum analyzer has correct trigger mode/level setting, otherwise you see lots of noise floor signal instead of WiFi packets. This is why you see few packet by luck (most of time it is the noise floor signal). Here is a spectrum setting from my side. As you can see I can capture the spectrum steadily.
[image]

(Above spectrum is with this fix: https://github.com/open-sdr/openwifi/issues/113#issuecomment-950631854 . Even without this fix, you should also see the spectrum if your spectrum analyzer trigger setting is correct)

JiaoXianjun, 2021-11-10:
You can debug further by:

1. Before running the injection program:
./sdrctl dev sdr0 set reg xpu 11 16 
./sdrctl dev sdr0 set reg xpu 1 4294967296
./sdrctl dev sdr0 set reg drv_tx 7 7

The 1st one turn off ACK TX.
The 2nd one turn off RX and CSMA controlling on TX
The 3nd one turn on the TX debug message in kernel

2. Run the packet inject program for 10 or 100 packets
3. Check the kernel debug message by running: dmesg

brealin, 2021-11-10:
When I use ad-hoc mode, I use UDP data to send packets to sdr0, and I can get the correct spectrum on the spectrum analyzer, so my current hardware environment should be no problem. 
[image]

I tried the above operation, but still nothing happens in the injection mode. 
[image]
[image]
[image]

brealin, 2021-11-11:
It has been verified that the 2.4G frequency WIFI can work normally. The default configuration in the package injection script is 44 channels of 5GHz frequency. The 5GHz signal of my board is too weak to trigger the signal analysis instrument. Thank you JiaoXianjun for your patient reply!

JiaoXianjun, 2021-11-12:
Good to know that the issue is not with openwifi.
I will close this issue.

closed 2021-11-12 by JiaoXianjun