open-sdr/openwifi #118: How can I set the protocol type of 802.11?

StoneHW, 2021-11-11:
1.How can I set the protocol type of 802.11? At present, hw_mode in the conf file can only support 802.11a/g, and 802.11n will be displayed as undefined. 
2.At the same time, if I want to set the 802.11g channel, there is no channel support, such as CHAN5G (64, 5320, IEEE80211_CHAN_RADAR) in ieee80211_channel channels_5GHz, the third option I understand should be a different protocol type, but in addition to 0 for 802.11a, Where is the specific setting location of IEEE80211_CHAN_RADAR?

JiaoXianjun, 2021-11-12:
1. The default fosdem.sh running in 11n mode. Please open that .sh file and find 11n hostapd example configuration file to read.
2. For the channel setting, we offer info in our project document: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#regulation-and-channel-config

We never try anything beyond above method. If you manage to extend it, would you please share with us how to to do it?

Thanks!

StoneHW, 2021-11-15:
I solved the first problem, but regarding the channel settings, it seems that the country does not have a unified adaptation. For example, the 32 and 34 channels in Japan cannot be supported. Is the channel selection here based on the Belgian channel regulations?
At the same time, I would like to ask, we provide the EDCA interface conf_tx, how to access and modify the EDCA command parameters, do you have any ideas?
Thank you for your answers.

JiaoXianjun, 2021-11-15:
Hello,

We never touch/set the country in the openwifi Linux/driver. You need to dig further on that.

We offered openwifi_conf_tx (in sdr.c) which is hooked to .conf_tx of ieee80211_ops.

You can check that function and do further experiment about EDCA. Currently that function is called by Linux, and Linux decides the EDCA parameters for each FPGA queue (of course the CSMA/CA engine in FPGA will do accordingly on each FPGA queue operation according to those parameters).

We don't know how Linux decides/sets those EDCA parameters. You need to dig further.

closed 2021-11-23 by StoneHW