open-sdr/openwifi #119: creating FPGA register inside SDR code

c2013hl, 2021-11-12:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

c2013hl, 2021-11-12:
Board: ADRV9361 (Zynq 7030) ADI SOM
We would like to create some registers inside the SDR code to capture some status for user level software to read. Would you give some advice how to proceed. We can code FPGA, user software code or modify the driver if needed. Hopefully, the driver already has a way to do this via IOCTL or some method. Thank you very much.

JiaoXianjun, 2021-11-12:
If you search register in the issues, you will find this answer:
https://github.com/open-sdr/openwifi/issues/32

JiaoXianjun, 2021-11-30:
If no further questions, I will close this issue.

c2013hl, 2021-11-30:
yes, you can close the issue

closed 2021-11-30 by JiaoXianjun