open-sdr/openwifi #120: FPGA Project building using TCL

waseem-arshad94, 2021-11-18:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

waseem-arshad94, 2021-11-18:
0.Yes I send you e-mail already in #103 

1. Default Image
2. Now I will modify the design.
3. Ubantu 18.04, Vivado 2019.1,2019_R1-2020_02_04.img.xz.
4. ADRV9361-Z7035 + ADRV1CRR-FMC I am using.
5. Will scan all channel in (2.4GHz,5Ghz).
6. At this stage I read all documents and related material, I build the FPGA design but I am unable to build using TCL.
7. I did this and working fine with default images.
8. I debug the projects and its working well, Now the time to build customized logic using openwifi project.
9. I will open the FPGA project using tcl and build the customized logic and will improve the project for future.

JiaoXianjun, 2021-11-18:
Hello,
What is your email address when you send email to me? Seems that I never receive it.

waseem-arshad94, 2021-11-18:
waseemarshad94@gmail.com

waseem-arshad94, 2021-11-18:
Dear Sir;

I did BS Electrical Electronics Engineering from a well reputed University
of Pakistan. Currently I am working as a Research Engineer in a well
reputed organization. I am implementing your openwifi project to receive
wifi bit-stream. Basically I want to scan all the wifi present in the
building. I read all the documents you provided me and learned a lot about
the project. I build an FPGA  project using the Linux environment. Now I
want to build the project using TCL. Now I need your help to implement and
contribute with you new things.

*Regards!*

*Waseem Arshad*
*Project Engineer, NRTC*
*Contact:+92-303-4459506*

JiaoXianjun, 2021-11-18:
I don’t understand what this issue is about. Can you elaborate further?

waseem-arshad94, 2021-11-19:
Hello;

**Done so far with openwifi?**

1. AP and Client SDR communicating successfully.
2. Scanning with one SDR all MAC addresses in the building.
3. FPGA default Block design in the vivado.

**Needs to be done**  

1. How openwifi.tcl will open?
2. I used the commands described about tcl project in openwifi-hw are not working.
3. I want to customized the FPGA logic for this purposes I need to opened openwifi.tcl

JiaoXianjun, 2021-11-19:
Could you read README of openwifi-hw repository?

waseem-arshad94, 2021-11-19:
Hi;
I read carefully but I am unable to open it.

JiaoXianjun, 2021-11-19:
In that readme, we ask for Vivado 2018.3 explicitly. So please make sure you use the correct version.

Then you should share with us:
1. The exact step by step operation you have done, including the command and parameter you use, software version, etc., better share the screenshot, etc.
2. The exact output after you run the command, especially the error message, etc.

Without above info, we can do nothing.

JiaoXianjun, 2021-11-30:
If no further question, I will close this issue.

waseem-arshad94, 2022-02-21:
Hi JiaoXianjun;
Sorry for late reply!
**I am using following setup**
**OS Version:** Ubantu 20.04
**Vivado Version:** 2018.3
**Command Running to build FPGA Design**

1. First I download the openwifi-hw  folder from github then,
2. Then I run, export XILINX_DIR=/home/ijaz/Vivado2018.3/ , After that I run, ./prepare_adi_lib.sh $XILINX_DIR
3.  BOARD_NAME=adrv9361z7035, then run script, ./prepare_adi_board_ip.sh $XILINX_DIR $BOARD_NAME
4. At this stage project reference design is build. After this 
5. I source VIVADO using command: source /home/ijaz/Vivado2018.3/Vivado/2018.3/settings64.sh
6. After this step I launch VIVADO and open the build reference block design
7. After opening the reference design I open TCL Console and run the following command 
8. soure ./openwifi.tcl and got error at this stage 
`source ./openwifi.tcl
# set  fd  [open  "openwifi_rev.coe"  w]
fatal: your current branch 'master' does not have any commits yet
    while executing
"exec git log -1 --pretty=%h"
    invoked from within
"set HASHCODE [exec git log -1 --pretty=%h]"
    (file "./openwifi.tcl" line 24)`

Thanks in advance!

waseem-arshad94, 2022-02-21:
Sorry 
**OS Version:** 18.04

JiaoXianjun, 2022-02-23:
"First I download the openwifi-hw folder from github then,"
Do you download the zip package of openwifi-hw or "git clone"?

JiaoXianjun, 2022-02-23:
I updated the building script a bit. You can try again.

waseem-arshad94, 2022-02-28:
Hi JiaoXianjun;
Your response is very appreciated! I tried it again and found this error in;
`ijaz@ijaz-HP:~/Documents/Project_Plan/openwifi-hw-master$ export XILINX_DIR=/home/ijaz/Vivado2018.3/
ijaz@ijaz-HP:~/Documents/Project_Plan/openwifi-hw-master$ ./prepare_adi_lib.sh $XILINX_DIR
$XILINX_DIR is found!
++ git submodule init adi-hdl
fatal: not a git repository (or any of the parent directories): .git
++ git submodule update adi-hdl
fatal: not a git repository (or any of the parent directories): .git
++ cd ./adi-hdl/library
++ git reset --hard f61d9707eb0a62533efd6facab59ab2444da94c9
fatal: not a git repository (or any of the parent directories): .git
++ cd /home/ijaz/Documents/Project_Plan/openwifi-hw-master`

waseem-arshad94, 2022-02-28:
Hi Jiao;
Your response is appreciated!
I run again the steps but now I found Error in very beginning after download new **openwifi-hw** again: 
`ijaz@ijaz-HP:~/Documents/Project_Plan/openwifi-hw-master$ export XILINX_DIR=/home/ijaz/Vivado2018.3/
ijaz@ijaz-HP:~/Documents/Project_Plan/openwifi-hw-master$ ./prepare_adi_lib.sh $XILINX_DIR
$XILINX_DIR is found!
++ git submodule init adi-hdl
fatal: not a git repository (or any of the parent directories): .git
++ git submodule update adi-hdl
fatal: not a git repository (or any of the parent directories): .git
++ cd ./adi-hdl/library
++ git reset --hard f61d9707eb0a62533efd6facab59ab2444da94c9
fatal: not a git repository (or any of the parent directories): .git
++ cd /home/ijaz/Documents/Project_Plan/openwifi-hw-master`
Thanks

JiaoXianjun, 2022-02-28:
“git submodule init adi-hdl fatal: not a git repository”

It is clear that you download our repo instead of git clone it. Please use “git clone” to clone our repo to your computer, then start from scratch.