open-sdr/openwifi #121: Is it possible to implement FTM protocal based on this project?

crownz-sec, 2021-11-21:
FTM protocal is used for ranging, details can be found here:https://www.researchgate.net/publication/328327674_Verification_Accuracy_Evaluation_of_WiFi_Fine_Time_Measurements_on_an_Open_Platform.
I just want to know is it possible to implement FTM protocal based on this project?

weiliu1011, 2021-11-21:
The referenced work uses Intel WiFi card, which does a lot of ranging calculation in the firmware. openwifi currently does not have these supported. 

In addition they also use a different Linux kernel than openwifi, in order to use the iwl driver, so some more challenges can appear.

In short this needs quite some time and effort.

JiaoXianjun, 2021-11-21:
Yes. It is possible. 

We have some results on synchronisation (not ranging, but basic concept is related). Checkout IEEE Transactions on Industrial Informatics. “Hardware Efficient Clock Synchronization across Wi-Fi and Ethernet Based Network Using PTP” on the publication page.

crownz-sec, 2021-11-22:
Thanks for kind reply!

closed 2021-11-22 by crownz-sec