open-sdr/openwifi #122: Timestamp of IQ capture program

gxhen, 2021-11-21:
Hi I am trying to analyze the timestamp returned by the IQ capture program. I use the MATLAB script 'test_iq_file_display.m' to analyze the saved 'iq.txt' file and visualize the timestamps. However, as shown in the figure below, I do not see the timestamp value gradually increasing. It fluctuates heavily and we cannot observe the trend.
[image]

However, the timestamps returned by the CSI collection program seems to be correct. The timestamp gradually increases with packet index which I presume is correct.
[image]

I am struggling to figure out the reason for the abnormal IQ capture timestamp, could you please give me some tips?

IQ capture settings and info:
1. board: zedboard + AD9361 and antsdr (both boards encounter the same problem)
2. iq_len_init: 4000
3. pre_trigger_len: 300
4. trigger condition: wh8d9 (receiver gives short preamble detected)

JiaoXianjun, 2021-11-22:
Hello,

It is due to 1 clk offset when we try to store the timestamp. Now I have fixed this issue.
You can take the fix from this branch: https://github.com/open-sdr/openwifi-hw/tree/iq_cap_timestamp_issue . You can see this issue and how I fix it in the latest commit in that branch.

To use this new FPGA code fix, you need to do:

1. Fetch the latest openwifi-hw including that branch, and switch to that branch in your local environment
2. Follow the section of "Modify IP cores:" in openwifi-hw README. (IP cores designed by verilog (rx_intf, xpu, etc). xpu as example:). Create the side_ch project, and export the side_ch IP to ip_repo/small/side_ch. -- I have done this for you in that fix, so you may skip this step. But it would be good for you to practice this step in case in the future you need to modify FPGA IP by yourself.
3. Re-build the whole FPGA design by following "Build FPGA:" section in openwifi-hw README. (till this step: ./sdk_update.sh $BOARD_NAME)
4. Now you need to follow https://github.com/open-sdr/openwifi#update-fpga to pick up the new FPGA bitstream, build the new BOOT.BIN and replace the BOOT.BIN on board.
5. Power cycle the board.
6. Now you should see correct timestamp while you do IQ capture.

Good luck!

gxhen, 2021-11-22:
Thanks very much for fixing this. I will try it according your instructions.

JiaoXianjun, 2021-11-26:
Have you tried it successfully?

gxhen, 2021-11-26:
Yes it works well. Thanks very much for your effort.

JiaoXianjun, 2021-11-26:
OK.
I will close this issue.

If there is any further problem, feel free to open a new issue.

closed 2021-11-26 by JiaoXianjun