open-sdr/openwifi #125: Can I change Txpower by set “AD9361_RADIO_ON_TX_ATT” in different value？

CaptainiFalcon, 2021-12-05:
It is very risky connecting tx to rx. It might damage your rx. 
You need a attenuator in between. Or lower the tx power.

The rf_init.sh only set the initial attenuation. The real-time attenuation is controlled in driver sdr.c. Please search set_tx_att in sdr.c. You will see how we do tx power control by Analog devices API.

_Originally posted by @JiaoXianjun in https://github.com/open-sdr/openwifi/issues/57#issuecomment-785682102_

CaptainiFalcon, 2021-12-05:
This issue is similar with issue#116（which I see after I ask the issue#125），*^____^*

JiaoXianjun, 2021-12-07:
> #define AD9361_RADIO_ON_TX_ATT 3000

#define AD9361_RADIO_ON_TX_ATT 3000 
will give you 3dB attenuation on TX. In other words, your TX power will be half as the original.

JiaoXianjun, 2022-04-01:
Now we have more easier way to configure the Tx ATT:
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#Tx-power-config