open-sdr/openwifi #126: when use cat  /pro/interrupts ,tx_itrpt didn't increased

XiaoYang980913, 2021-12-05:
AP-client mode 
when use cat  /pro/interrupts ,tx_itrpt didn't increased,and the number of tx_itrpt  is not equal to xilinx-dma-controller.
At this time, ILA was used for loopback test, and it was found that client test did not send anything, while AP test was still sending andnormally  
Why does this happen? Initially, I think cilent card is in a certain state of TX_control in XPU and cannot exit. I wonder whether such a situation exists ?

JiaoXianjun, 2021-12-05:
Without the info, that you need to offer, in the issue template, we hardly can help.

XiaoYang980913, 2021-12-05:
thanks
1. image:openwifi-1.2.1-leuven-2-32bit.img
2. modification: the length of CP (ofdm_tx.v&ofdm_rx.v)，Changed the value of XPU register 11,17,18
3. BOARD type: zc706+9361
4.channel num: 44
5. operating steps:
AP: 
ssh root@192.168.10.122
(password: openwifi)
cd openwifi
./fosdem-11ag.sh
cat /proc/interrupts

message:Tx_itrpt grows normally and is equal to the Xilinx-DAM-Controller number  

Before changing the XPU reg:
client:
ssh root@192.168.10.122
(password: openwifi)
service network-manager stop
cd openwifi
./wgd.sh
ifconfig sdr0 up
iwlist sdr0 scan
(After this command, I can see my AP device,Each time I use it, I can see that my tx_itrpt and rx_pkt_intr grow normally )
Then i use wpa_supplicant -i sdr0 -c wpa-openwifi.conf
(Successfully initialized wpa_supplicant   
 sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED
sdr0: CTRL-EVENT-SCAN-STARTED...........)
cat /proc/interrupts
(At this time, tx ITRPT and XilinX-DMA-Controller values are inconsistent)
Then i use iwlist sdr0 scan
(After this command, I can't see my AP device,and I can't see my tx_itrpt and rx_pkt_intr grow)

Change the time-related register in XPU (11 17 18) ,then found with the above situation
ILA was used to observe the data, and rx_intf reg 3 was changed to observe the data directly transmitted from TX_INTF to RX_INTF. It was found that the data was decoded, FCS output was normal, and PHY worked normally.

JiaoXianjun, 2021-12-05:
Since it is not our image, and we don't have your modification, it is difficult for us to help.

No tx_itrpt means this signal https://github.com/open-sdr/openwifi-hw/blob/70d7b29e87866b7e51324ed547fb6d75725cf38d/ip/tx_intf/src/tx_intf.v#L60 is not raised. You need to trace why it is not raised by inserting ila to all related signals.

XiaoYang980913, 2021-12-05:
OK，thanks a lot

XiaoYang980913, 2021-12-08:
Sorry to bother you again
I wonder if the SDR error message is displayed like:
sdr,sdr openwifi_tx: WARNING ieee80211_stop_queue prio 0 queue 0 no room flag 0 hw queue len 0013003d request 12 wr 13 rd 15

Conditions for printing this error message: "check whether FPGA dma fifo and queue (queue_idx) has enough room"
**I observed the correct print message wr_idx is usually one bigger than rd_idx,but at this point my rd_idx is bigger than wr_idx**,.And my no room flag is still 0,check the register of tx_intf at this time, four queues FIFO are not full.

I would like to know what parts I need to check in case of this error output, thank you very much

closed 2021-12-12 by XiaoYang980913