open-sdr/openwifi #130: Pilot signal and Long training symbol

Eugene-Liu-milkshake, 2021-12-15:
From Eugene Liu:

 Hi, I am looking to do some experiments with SDR using openwifi on Analog Device adrv9364z7020 hardware. 

I am studing channel estimate. From the csi document, 11a/g only 48 subcarriers valid.

Could we obtain pilot signal or LTS from openwifi ? 

Thanks for your openwifi project, this quite helpful for us to improve our research.

Sincerely,
Eugene Liu.

JiaoXianjun, 2021-12-15:
CSI data actually is frequency domain modulation-removed LTS, that has only pilots, not data. So we already have it.

What do you mean by "obtain". We have I/Q capture app notes that allow you to obtain everything:
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq.md
https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/iq_2ant.md

Eugene-Liu-Incubator, 2021-12-18:
From Eugene Liu:

Thanks for your reply. I have clearly understood the relationship of CSI and pilot. 

The 'obtain' means is there any methods to get Long training symbols value from openwifi ?
I haved read the document of IQ mode. 
If I change the trigger condition to parameter 8, which means receiver gives long preamble detected.
What would it cause to the receving data from iq_capture.py after parsing ?

Sincerely,
Eugene Liu.

JiaoXianjun, 2021-12-18:
Yes you can capture I/Q sample of the packet including LTS. 

Just try it and see what happen.

Eugene-Liu-Incubator, 2021-12-19:
Got it. I will give it a try, thank you.

Can I set target/source MAC address as capture condition when running I/Q sample ? like csi capture does
The command ./side_ch_ctl wh1h4001 still work  at I/Q sample ?

In addition, our group found that when receiving equalizer (hostapd set as 802.11a), the first column always perform BPSK.
Is that SIGNAL field ?

Sincerely,
Eugene Liu.

closed 2022-04-11 by Eugene-Liu-Incubator