open-sdr/openwifi #131: iq_capture app

Shangjie-Yang, 2021-12-23:
Hi,
    I am having the following problem with the iq_capture tool.
    I send a continuous IQ signal and use the iq_capture tool to capture it, but each time I capture 4094 symbols it interrupts once, and after the interruption there is a delay before continuing the capture, and the delay is not fixed.
    Then I modified the 'wh11d4094' instruction to 'wh11d8187',  after the modification, the phenomenon became that there was an interrupt after every 8187 symbols were captured, and the interrupt generated a time delay and caused some data to be missed.
    I would like to know the reason for the delay,and what is the way to avoid these delays, or make these delays into a fixed value.
    Thanks.
[image]

JiaoXianjun, 2021-12-28:
Can you share me the full command sequence? Then I will try to reproduce it.

JiaoXianjun, 2021-12-28:
Can you also answer all questions in our issue template? Those information are very important for us to understand the issue.

closed 2022-04-13 by JiaoXianjun

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.