open-sdr/openwifi #134: MAC address matching when using the CSI collection tool

gxhen, 2022-01-08:
That's indeed a great project. Thanks for your efforts on this. I have been using your CSI collection tool for a while but encountered a problem recently.

I was running openwifi in the monitor mode on channel 11 (2.4 GHz band), just like the example you provided in the app_notes. I have two other COTS wi-fi devices, one configured as AP and another as STA. The STA sends packets to the AP continuously in a 0.1 second interval, and AP sends acknowledgements back. We aim to use openwifi to capture the packets from STA and extract their CSI.

I config the capture condition as addr2 matching (wh1h4001). However, it seems that a large number of packets are lost. The side info count increases by one every 2 or 3 seconds, which is significantly larger than the actual packet interval of 0.1s. I confirm the AP and STA are running properly by monitoring them in Wireshark. In addition to this, the packet loss can be improved by turning off the conditional capturing (wh1h0001). In this command the packets are smoothly collected and side info count increases very fast. There are not too many other wifi devices in channel 11 therefore the collected packets are all from our AP/STA (if I turn off my STA the side info count stops increasing again).

Other info: We directly use your image without any modification. We tried on both Zedboard and ANTSDR but the problem was not improved. The experiments are conducted on Wi-Fi channel 11 (2.4 GHz frequency band).

I need your kind help with two questions:
1) Whether the packet loss is caused by the insufficient ARM core on Zedboard and ANTSDR? In other words, are they unable to quickly parse MAC address? Will a higher-end board improve this?
2) Is it possible to return addr2 together with CSI/frequency offset/equalizer output via ssh by modifying the openwifi driver?

Thanks in advance.

JiaoXianjun, 2022-01-27:
Can you run tcpdump on openwifi board (monitor the channel your traffic is in) to check how many packet received?

ymature, 2022-02-18:
Then Mr. Jiao, I need your help with following question:
  How to run tcpdump on openwifi board ?

Thanks in advance.

JiaoXianjun, 2022-02-22:
tcpdump is installed on openwifi board by default. You can use it on board directly.

closed 2022-04-13 by JiaoXianjun