open-sdr/openwifi #135: After modifying the driver, follow the "Update Driver" operation in the README file and the following problems occur?

maykolo, 2022-01-26:
The first time, I downloaded all the files, and then modified the places I need to modify, and then there was a problem with the compilation.
The second time, I reverted to the original file, i.e., without modifying the source code, and then compiling also had a relative problem.
So this should not be an error caused by the code I modified.
Environment: "openwifi-1.2.0" and "linux-2019_R1"
Compilation error:
`openwifi/driver/git_rev.h:1:17: error: invalid suffix "x" on integer `constan’`
 #define GIT_REV 0x`

[image]

JiaoXianjun, 2022-01-26:
What is your OS version and Vivado version?

maykolo, 2022-01-26:
Vivado2018.3

JiaoXianjun, 2022-01-26:
OS version

maykolo, 2022-01-26:
Ubuntu16.04&nbsp; Vivado2018.3

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Jiao ***@***.***&gt;; 
发送时间: 2022年1月26日(星期三) 下午5:45
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [open-sdr/openwifi] After modifying the driver, follow the &quot;Update Driver&quot; operation in the README file and the following problems occur? (Issue #135)

操作系统版本

—
直接回复此邮件，在 GitHub 上查看或者取消订阅.
GitHub Mobile 上的分流通知IOS 系统或机器人. 
你收到了这是因为你是这个线程的作者。留言 ID ***@***.***与&gt;

JiaoXianjun, 2022-01-26:
what is the content of git_rev.h?

JiaoXianjun, 2022-01-26:
what command you have run to produce that error?

maykolo, 2022-01-26:
There is only one line of code in this file。
is #define GIT_REV 0x

I ran "./make_all.sh $XILINX_DIR ARCH_BIT"

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Jiao ***@***.***&gt;; 
发送时间: 2022年1月26日(星期三) 下午5:49
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [open-sdr/openwifi] After modifying the driver, follow the &quot;Update Driver&quot; operation in the README file and the following problems occur? (Issue #135)

你运行了哪个命令来产生这个错误?

—
直接回复此邮件，在 GitHub 上查看或者取消订阅.
GitHub Mobile 上的分流通知IOS 系统或机器人. 
你收到了这是因为你是这个线程的作者。留言 ID ***@***.***与&gt;

JiaoXianjun, 2022-01-26:
if your directory is not git repository, you need to have a fake git_rev.h. Or put some fake revision number to this line (make_all.sh):
```
echo "#define GIT_REV 0x"$(git log -1 --pretty=%h) > git_rev.h
```

maykolo, 2022-01-26:
What does this mean and how to add it?

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Jiao ***@***.***&gt;; 
发送时间: 2022年1月26日(星期三) 下午5:51
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [open-sdr/openwifi] After modifying the driver, follow the &quot;Update Driver&quot; operation in the README file and the following problems occur? (Issue #135)

如果你的目录不是 git 仓库,你需要有个假的 git _ rev . h 。或者在这一行添加一些假的版本号( make _ all . sh ):
echo "# define GIT _ REV 0x "$( git log - 1 - pretty =% h )&gt; git _ rev . h
—
直接回复此邮件，在 GitHub 上查看或者取消订阅.
GitHub Mobile 上的分流通知IOS 系统或机器人. 
你收到了这是因为你是这个线程的作者。留言 ID ***@***.***与&gt;

JiaoXianjun, 2022-01-26:
You can just remove that line, and create your own fake git_rev.sh with content:
#define GIT_REV 0x00000

maykolo, 2022-01-26:
Hello, according to your suggestion, I modified the "0x" inside to "0x00000" and recompiled, but it changed back to "0x"

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Jiao ***@***.***&gt;; 
发送时间: 2022年1月26日(星期三) 下午5:56
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [open-sdr/openwifi] After modifying the driver, follow the &quot;Update Driver&quot; operation in the README file and the following problems occur? (Issue #135)

你可以删除这一行,创建自己的假 git _ rev . sh ,其中包含:
# define GIT _ REV 0x00000

—
直接回复此邮件，在 GitHub 上查看或者取消订阅.
GitHub Mobile 上的分流通知IOS 系统或机器人. 
你收到了这是因为你是这个线程的作者。留言 ID ***@***.***与&gt;

JiaoXianjun, 2022-01-26:
remove that line from your make_all.sh!

maykolo, 2022-01-26:
Thanks, I got it done.

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Jiao ***@***.***&gt;; 
发送时间: 2022年1月26日(星期三) 晚上6:00
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [open-sdr/openwifi] After modifying the driver, follow the &quot;Update Driver&quot; operation in the README file and the following problems occur? (Issue #135)

从你的 make _ all . sh 中删除这一行!

—
直接回复此邮件，在 GitHub 上查看或者取消订阅.
GitHub Mobile 上的分流通知IOS 系统或机器人. 
你收到了这是因为你是这个线程的作者。留言 ID ***@***.***与&gt;

closed 2022-01-26 by maykolo