open-sdr/openwifi #138: modulation.v, such as 16'hBEA0, how does this BEA0 come from?

peking428, 2022-02-19:
I don't understand the data under 64QAM in the code file modulation.v, such as 16'hBEA0, how does this BEA0 come from? How to calculate it?

JiaoXianjun, 2022-02-22:
You could convert those hex number to signed 16bit integer and plot them in the constellation square, then I believe you will understand.

peking428, 2022-02-25:
thank you!

---- 回复的原邮件 ----
| 发件人 | Jiao ***@***.***> |
| 日期 | 2022年02月22日 23:22 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [open-sdr/openwifi] modulation.v, such as 16'hBEA0, how does this BEA0 come from? (Issue #138) |

You could convert those hex number to signed 16bit integer and plot them in the constellation square, then I believe you will understand.

closed 2022-04-01 by JiaoXianjun