open-sdr/openwifi #139: How to get timestamp of WiFi packets in the way of this figure?

GuanLiliang, 2022-02-26:
board: Antsdr e310
I'm sorry to trouble you again.
Actually, I succeed to get IQ information by the steps in https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md
Now what I want to do is to map CSI(IQ information) to the WiFi packets.
You have said that I can match the timestamp (TSF timer value) between WiFi packet and CSI information, because this is the unique same identity of a WiFi packet and related CSI information.
But I can't get the timestamp based on the same TSF timer(time base), because I run Wireshark in PC and run side_ch_ctl in board.
Do you mean that I must get WiFi packets just like this figure rather than running Wireshark in PC?
[image]
I don't know how to get WiFi packets in this way.
I also wanna know that if I can get CSI information and WiFi packets on board at the same time.
In facts, I want to use CSI information to do side channel analysis of WiFi packets.

JiaoXianjun, 2022-02-26:
You need to run tcpdump or wireshark (tshark?) or other packet sniffer on board to get the timestamp of WiFi packet. How to install them on board? Please learn that from the script post_config.sh.

of course you the csi is firstly available on board, then it is transferred to pc. Please learn the program (source code) we run on board to learn how to get csi on board.

JiaoXianjun, 2022-02-26:
Openwifi report the TSF timestamp of rx packet to Linux in the openwifi_rx_interrupt() function of SDR.c. Please learn that function’s source code.

GuanLiliang, 2022-02-27:
when I run tcpdump in board.

JiaoXianjun, 2022-02-27:
yes

closed 2022-04-13 by JiaoXianjun