open-sdr/openwifi #140: The tx power of openwifi based on antsdr

black-pigeon, 2022-02-28:
1. Our image is used directly or you build your own image?
     yes
2. What is your own modification?
     add rf switch control for antsdr. I can control the rf switch gpio in the linux running on  antsdr.
3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
    same as the master branch
4. Board/hardware type
    antsdr
5. WiFi channel number
    5GHz channel 44
6. result
   when I power the antsdr and running openwifi. I connect the output of antsdr to a spectrum, the default result shows a ultra week signal at 5.229GHz.
[image]
   Then I try to change the  [tx Attenuation](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#tx_attenuation_control), I got the result that the tx attenuation is 0dB of TX1 ,then I set the TX2 attenuation to -10dB. But this can't improve the performance.
```bash
root@analog:~# cat /sys/bus/iio/devices/iio\:device1/out_voltage0_hardwaregain
0.000000 dB
root@analog:~# cat /sys/bus/iio/devices/iio\:device1/out_voltage1_hardwaregain
-10.000000 dB
root@analog:~# 
```

  Then I try to send a [bits_tone](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#bist_tone), and I got a bits_tone signal transmitted by the ad9361. The power of bits_tone is about -5dBm which seems performed well. The marker 2 is  bist_tone, and the marker 1 is openwifi default output.
[image]

could you please tell me, how to make a openwifi transmit some signnal? The openwifi signal is weak, I hope I can improve this. Some customers have the same problem with antsdr.

HavingaThijs, 2022-02-28:
I have used the antsdr before at 5GHz without RF switch. 
It looks like the spectrum from your first image is not the openwifi spectrum, but some leakage. You can use the [packet injection application note](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/inject_80211.md) to make openwifi transmit some signal. You can also check this issue #117 for some details on how to capture it using a spectrum analyzer.

black-pigeon, 2022-02-28:
Thanks for your replay.
[image]

I can trigger the output of the antsdr after I type following command.  The result is close to  [this issue](https://github.com/open-sdr/openwifi/issues/117#issuecomment-964200627). but when antsdr work at ap mode, why performance is not that good?
```bash
./inject_80211/inject_80211 -m n -r 2 -n 10000 -s 1024  -d 40000 sdr0
```

HavingaThijs, 2022-02-28:
In AP mode, when not connected to any client, it sends only some beacons periodically (I think the default is every 0.1s). These packets are also not so big and may therefore be hard to capture. When you set the trigger power level (and maybe sweep time) correctly, you should also see something like this in AP mode.

black-pigeon, 2022-03-01:
In the AP mode, I can also capture the packet, I set the trigger condition to -30dBm and it will trigger.
However, when I use my PC to connect the openwifi running on antsdr, the distance between the antsdr and my PC is about 1m, the link info of my pc shows as follow.
[image]

When I move very close to antsdr the result is shown as follow.
[image]

Have you tested how far openwifi can transmit before? How about the performance of FMCOMMS or ADRV9361/4 ?

black-pigeon, 2022-03-02:
I followed this [issue](https://github.com/open-sdr/openwifi/issues/113#issuecomment-950631854) to recompile the driver, and copied all the files   changed to the board.
On the board, I used following command to run openwifi:
```bash
~/openwifi/wgd.sh
~/openwifi/fosdem-11ag.sh
```
In the spectrum, I turned the trace on and set the max hold, then  I using trigger to capture signal.
I got a result of the output of antsdr, which is  similar to [issue 117](https://github.com/open-sdr/openwifi/issues/117)
[image]

HavingaThijs, 2022-03-03:
The spectrum indeed looks ok. 
We never really tested how far it can actually transmit, but it definitely varies per board/antenna configuration etc. The plan is to create a next release really soon, coming with several improvements that might influence this. After this, we can do some tests.

black-pigeon, 2022-03-04:
Waiting for your update. Then I can commit the the rf switch control for antsdr (ง •̀_•́)ง !!

black-pigeon, 2022-03-24:
When we add an external PA to the ANTSDR, the performance is improved a lot. 

So I think it's better to add a PA to the ANTSDR in the next hardware release. I will close this issue, thanks for your kind work.

closed 2022-03-24 by black-pigeon