open-sdr/openwifi #142: Customization in openwifi fpga design

waseem-arshad94, 2022-03-01:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

waseem-arshad94, 2022-03-01:
0.Yes I send you e-mail already in https://github.com/open-sdr/openwifi/issues/103

1. I will build own image 
2. Now I will modify the FPGA design.
3. Ubantu 18.04, Vivado 2019.1,2019_R1-2020_02_04.img.xz.
4. ADRV9361-Z7035 + ADRV1CRR-FMC I am using.
5. Will scan all channel in (2.4GHz,5Ghz).
6. At this stage I read all documents and related material, I build the FPGA design. I am scanning all wifi channels. After scanning I receive complete list of MAC addresses which are monitoring by Terminal. My question is: "**If there is another wifi router apart from these, there should be a TTL high logic from HDL design?**" 
7. I burn default images from the table and they work fine for me and debug analysis were good. Scanning is perfect.
8. default images works good.
9. Please provide **openwifi_ip** documentation helps me to understand better.

JiaoXianjun, 2022-04-01:
Please follow https://github.com/open-sdr/openwifi-hw#readme

closed 2022-04-01 by JiaoXianjun