open-sdr/openwifi #143: New board SDRPi request a branch.

HexSDR, 2022-03-11:
Hi ,
I have designed a zynq + ad9361 board which runs openwifi well .Hareware feature is : ZYNQ XC7Z020CLG400 ,1GB DDR3 memory fo PS, 1G ETHER RJ45 for PS,1G  ETHER RJ45 for PL, USB OTG(act as USB host), USB uarts for PS and PL.on board USB to JTAG debuger,TF card , bootable QSPI FLASH and also 27 IO from PL bank; AD9361 RF design is based FMCOMMS3 with TX amplifier. 

These features listed above is similar as ZEDBOARD+FMCOMMS3.For the applications as CSI data dumping ,I also designed an UDP stack accompanied by this PL ethernet port on board .Using this method I will try to modify some source code to ofload the bandwidth and CPU usage,making a more clear and smart way to capture raw IQ data from RF chip .

This SDRPi which size is as small as  Raspberry Pi runing openwifi can be embedded in user application such as UAV,IoT,WIFI router... I request a new branch for this board named as SDRPi.
Thanks.

JiaoXianjun, 2022-03-11:
Looks promising!
Already manufactured? Could you attach a photo?

We are preparing a big release. Will be public soon.
After that, I will create a branch for your contribution.

HexSDR, 2022-03-14:
This is the previous version of a demo board. This current board  modified slightly in progressing will be available in about one month.

[image]

[image]

JiaoXianjun, 2022-04-01:
Hello,

We have just published our new openwifi release openwifi-1.3.0-wilsele.
I also created sdrpi branch for openwifi and openwifi-hw:
https://github.com/open-sdr/openwifi/tree/sdrpi
https://github.com/open-sdr/openwifi-hw/tree/sdrpi

You can put your changes needed by sdrpi board by creating pull request from your own branch (please create) to sdrpi branch. Then we can review and discuss on sdrpi branch.

JiaoXianjun, 2022-04-01:
More important, before contribution, could you please check https://github.com/open-sdr/openwifi/blob/master/CONTRIBUTING.md

and sign the CLA. Then send it to Filip.Louagie@UGent.be, xianjun.jiao@ugent.be and Ingrid.moerman@ugent.be ?

Thanks!

HexSDR, 2022-04-03:
Thanks ,
I have signed this CLA and mailed to you.  I will upload my design later.

catkira, 2022-04-10:
SDRPi looks nice, is it already available somewhere? Is it commercial or open source?

JiaoXianjun, 2022-04-10:
Indeed, if there is website/github for sdrpi, where people could check schematic/pcb design, that would be great.

HexSDR, 2022-05-09:
It has already been designed OK and  now under producting ,for some reasons the producting processing is so slow ,I am very sorry for this delay.  
A github for sdrpi is under developing ,some more thing will put on it . and also will update to this sdrpi branch. In my opinion, it may not open all source files , but will absolutely be sufficent for sdrpi user .

JiaoXianjun, 2022-08-08:
@HexSDR I have checked https://github.com/HexSDR/openofdm_rx/tree/dot11zynq
It is OK.
Would you please make a pull request to https://github.com/open-sdr/openofdm/tree/dot11zynq ?

Thanks!

HexSDR, 2022-08-08:
OK ， I will try to do it,thanks