open-sdr/openwifi #144: No Beacon frames on 2.4GHz channels when in monitor mode

pzanna, 2022-03-13:
I'm using a adrv9361z7035 with the latest image (openwifi-1.2.1-leuven-2-32bit.img). I boot the board and run "./wgd.sh" to load the drivers and then run "./sdrctl dev sdr0 set reg drv_rx 7 3" to enable debug messages.

If I enable monitor mode on 5GHz (./monitor_ch.sh sdr0 36), when there is another AP on the same channel, I see hundreds of beacon frames such as "00:05:33.069144 174374521us tsft 6.0 Mb/s 5180 MHz 11a -49dB signal antenna 0 Beacon (XXXXXX) [6.0* 9.0 12.0* 18.0 24.0* 36.0 48.0 54.0 Mbit] ESS CH: 36, PRIVACY" when using "tcpdump -i sdr0 type mgt subtype beacon".

However, if I change monitor mode to 2.4GHz (./monitor_ch.sh sdr0 11) which also has an AP on the same channel, tcpdump shows no beacon frames.

I have also noticed that all frames on 2.4GHz show bad fcs such as this "00:17:55.930850 61531040us tsft bad-fcs 6.0 Mb/s 2462 MHz 11g -92dB signal antenna 0 Probe Response[|802.11]" where 5GHz frames do not.

JiaoXianjun, 2022-03-14:
it might due to some issue we are working on.
2.4GHz is quite busy, and some times the receiver even is busy with "fake" signal (bad fcs). in this case, beacon frames might be missed (receiver is busy with some fake signal).

what about you put the AP very close to the board?

what is the result of "iwlist sdr0 scan"?

We will release a new version soon, where the issue will be solved.

JiaoXianjun, 2022-03-24:
Tow SDRs (one client, one AP) can work in 2.4GHz and communicate with each other without issue.

JiaoXianjun, 2022-03-24:
we are preparing new releases for this.

JiaoXianjun, 2022-04-01:
Could you try the new release?

pzanna, 2022-04-02:
I have tried the new release and the outcome is the same.

JiaoXianjun, 2022-04-02:
Please follow this https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#receiver-sensitivity-control to make your receiver less sensitive. For instance, set the threshold to 70 or 60. 

Share me the dmesg output when you have set drv_rx 7 15.

pzanna, 2022-04-04:
### On 2.4GHz I get:

sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 32B ht0aggr0/0 sgi0 240M FC0094 DI0000 ADDRa8db031b1b02/78d38ddffd1a/0500f050ffff SC4095 fcs1 buf_idx49 -43dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI0000 ADDR78d38ddffd1a/000000000000/000000000000 SC0 fcs1 buf_idx50 -63dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI012c ADDRa8db031b1b02/000000000000/000000000000 SC0 fcs1 buf_idx51 -42dBm
sdr,sdr openwifi_rx: 28B ht0aggr0/0 sgi0 240M FC0148 DI0158 ADDR78d38ddffd1a/a8db031b1b02/78d38ddffd1a SC1265 fcs1 buf_idx52 -62dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI012c ADDRa8db031b1b02/a8db031b1b02/78d38ddffd1a SC1265 fcs1 buf_idx53 -42dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 30B ht0aggr0/0 sgi0 60M FC01c8 DI003c ADDR78d38ddffd1a/c8f7333ea766/78d38ddffd1a SC0 fcs1 buf_idx54 -49dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 60M FC00d4 DI0000 ADDRc8f7333ea766/c8f7333ea766/78d38ddffd1a SC0 fcs1 buf_idx55 -41dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00c4 DI0072 ADDRa8db031b1b02/000000000000/000000000000 SC0 fcs1 buf_idx56 -41dBm
sdr,sdr openwifi_rx: 50B ht0aggr0/0 sgi0 240M FC92ef DI1c79 ADDR769831e75367/8c1837c00897/0d8c7a2680bf SC1680 fcs0 buf_idx57 -41dBm
sdr,sdr openwifi_rx: 106B ht1aggr0/0 sgi1 585M FC099d DI49ac ADDR3e700493c5a6/c25253173e3a/c0247646f0f9 SC2794 fcs0 buf_idx58 -43dBm
sdr,sdr openwifi_rx: 106B ht1aggr0/0 sgi1 260M FC4a88 DI002c ADDRa8db031b1b02/78d38ddffd1a/78d38ddffd1a SC1049 fcs1 buf_idx59 -43dBm
sdr,sdr openwifi_rx: 106B ht1aggr0/0 sgi1 260M FC4a88 DI002c ADDRa8db031b1b02/78d38ddffd1a/78d38ddffd1a SC1049 fcs1 buf_idx60 -46dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI0000 ADDR78d38ddffd1a/78d38ddffd1a/78d38ddffd1a SC1049 fcs1 buf_idx61 -62dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 28B ht0aggr0/0 sgi0 240M FC1148 DI0158 ADDR78d38ddffd1a/a8db031b1b02/78d38ddffd1a SC1268 fcs1 buf_idx62 -62dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI012c ADDRa8db031b1b02/a8db031b1b02/78d38ddffd1a SC1268 fcs1 buf_idx63 -42dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 28B ht0aggr0/0 sgi0 240M FC0148 DI0158 ADDR78d38ddffd1a/a8db031b1b02/78d38ddffd1a SC1269 fcs1 buf_idx0 -60dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI012c ADDRa8db031b1b02/a8db031b1b02/78d38ddffd1a SC1269 fcs1 buf_idx1 -44dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI0000 ADDR78d38ddffd1a/000000000000/000000000000 SC0 fcs1 buf_idx2 -62dBm
sdr,sdr openwifi_rx: 20B ht0aggr0/0 sgi0 240M FC00b4 DI009a ADDR78d38ddffd1a/a8db031b1b02/000000000000 SC0 fcs1 buf_idx3 -64dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00c4 DI006e ADDRa8db031b1b02/a8db031b1b02/000000000000 SC0 fcs1 buf_idx4 -42dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 20B ht0aggr0/0 sgi0 240M FCbe1c DI3b8e ADDR86a3c24cbd7e/dcb63330fc2f/000000000000 SC0 fcs0 buf_idx5 -63dBm
sdr,sdr openwifi_rx: 20B ht0aggr0/0 sgi0 240M FCbe1c DI3b8e ADDR86a3c24cbd7e/dcb63330fc2f/000000000000 SC0 fcs0 buf_idx6 -63dBm
sdr,sdr openwifi_rx: 20B ht0aggr0/0 sgi0 120M FC00b4 DI00b6 ADDR78d38ddffd1a/a8db031b1b02/000000000000 SC0 fcs1 buf_idx7 -62dBm
sdr,sdr openwifi_rx: 20B ht0aggr0/0 sgi0 120M FC00b4 DI00b6 ADDR78d38ddffd1a/a8db031b1b02/000000000000 SC0 fcs1 buf_idx8 -62dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 120M FC00c4 DI0086 ADDRa8db031b1b02/a8db031b1b02/000000000000 SC0 fcs1 buf_idx9 -42dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 32B ht0aggr0/0 sgi0 120M FC0094 DI0000 ADDRa8db031b1b02/78d38ddffd1a/05001051ffff SC4095 fcs1 buf_idx10 -42dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 60M FC10c4 DI0edd ADDRe03e4408551e/000000000000/000000000000 SC0 fcs1 buf_idx11 -67dBm
sdr,sdr openwifi_rx: 28B ht0aggr0/0 sgi0 60M FC0148 DI003c ADDR78d38ddffd1a/f4cfa2198bff/78d38ddffd1a SC1645 fcs1 buf_idx12 -53dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 60M FC00d4 DI0000 ADDRf4cfa2198bff/000000000000/000000000000 SC0 fcs1 buf_idx13 -41dBm
sdr,sdr openwifi_rx: 80B ht0aggr0/0 sgi0 480M FC4108 DI003c ADDR78d38ddffd1a/f4cfa2198bff/ffffffffffff SC2992 fcs1 buf_idx14 -54dBm
sdr,sdr openwifi_rx: 28B ht0aggr0/0 sgi0 60M FC1148 DI003c ADDR78d38ddffd1a/f4cfa2198bff/78d38ddffd1a SC1646 fcs1 buf_idx15 -52dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 60M FC00d4 DI0000 ADDRf4cfa2198bff/000000000000/000000000000 SC0 fcs1 buf_idx16 -40dBm
sdr,sdr openwifi_rx: 30B ht0aggr0/0 sgi0 60M FC01c8 DI003c ADDR78d38ddffd1a/c8f7333ea766/78d38ddffd1a SC0 fcs1 buf_idx17 -48dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 60M FC00d4 DI0000 ADDRc8f7333ea766/000000000000/000000000000 SC0 fcs1 buf_idx18 -41dBm
sdr,sdr openwifi_rx: 126B ht1aggr0/0 sgi1 390M FC4188 DI002c ADDR78d38ddffd1a/c8f7333ea766/78d38ddffd1a SC2666 fcs1 buf_idx19 -49dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI0000 ADDRc8f7333ea766/c8f7333ea766/78d38ddffd1a SC2666 fcs1 buf_idx20 -44dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
[... cut ...]
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 28B ht0aggr0/0 sgi0 240M FC570c DI06fb ADDRcba1539d08be/10984a770f56/5a70e65ca422 SC2147 fcs0 buf_idx16 -61dBm
sdr,sdr openwifi_rx: 14B ht0aggr0/0 sgi0 240M FC00d4 DI0000 ADDRc869cd0c9552/10984a770f56/5a70e65ca422 SC2147 fcs1 buf_idx17 -49dBm
sdr,sdr openwifi_rx: WARNING loop_count 2
sdr,sdr openwifi_rx: WARNING loop_count 0
sdr,sdr openwifi_rx: 341B ht0aggr0/0 sgi0 120M FC00d0 DI0000 ADDRffffffffffff/a67746bd44a4/002500ff9473 SC1798 fcs1 buf_idx18 -54dBm

JiaoXianjun, 2022-04-04:
Those fffffff should be beacon?
Can you set that AP in 20MHz mode (a or n) and make sure disable 11b?

what about “iwlist sdr0 scan”

JiaoXianjun, 2022-04-13:
Did you figure out the reason?

closed 2022-09-29 by JiaoXianjun