open-sdr/openwifi #147: openwifi IP can be used in reference design?

waseem-arshad94, 2022-03-22:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

waseem-arshad94, 2022-03-22:
0.Yes I send you e-mail already in https://github.com/open-sdr/openwifi/issues/103

1. Default Image
2. Now I will modify the design.
3. Ubantu 18.04, Vivado 2019.1,2019_R1-2020_02_04.img.xz.
4. ADRV9361-Z7035 + ADRV1CRR-FMC I am using.
5. Will scan all channel in (2.4GHz,5Ghz).
6. At this stage I read all documents and related material, I build the FPGA design successfully, and  Now I want to use openwifi IP in other reference design.
7. I did this and working fine with default images.
8. I debug the projects and its working well, Now the time to build customized logic using openwifi project.
9. I will open the FPGA project using tcl and build the customized logic and will improve the project for future.

waseem-arshad94, 2022-03-22:
Dear Sir;
When I start scanning the WiFi routers in the building then openwifi design scan successfully all routers present in the building.
I have few questions about openwifi design can you help me.

1. Can I modify the openwifi hdl logic in other reference design like **2019_r2 HDL reference design.**  
2. When I scan building routers it scan whole routers. Can I stop to scan the scan the some routers means for example I have 50 routers in my building I want to allow the openwifi should be scan only friendly routers **let say 30 should be scan and 20 non-friendly should not be scan is possible?**
3. Any script or logic that can differentiate friendly or non-friendly WiFi routers.

Thanks

JiaoXianjun, 2022-04-01:
1. Yes follow our openwifi-hw README https://github.com/open-sdr/openwifi-hw#readme , you can change openwifi hdl logic

For 2&3, they are common questions for any wifi, because those are triggered by high level Linux operations. openwifi, just like any other wifi chip, only offers driver and chip (FPGA). For the control functionalities above driver, it runs out of openwifi control. You need to seek some common high level solution on internet.

closed 2022-04-13 by JiaoXianjun