open-sdr/openwifi #150: Unable to successfully receive csi information

czmxiaoming, 2022-03-29:
Hello, I have a question I would like to ask. I have one antsdr board.When I'm running the csi section,I can see on board outputs like:
loop 64 side info count 62
loop 128 side info count 125
loop 192 side info count 189
......
 but the computer sometimes can't properly generate the file named side_info.txt , sometimes the file with a capacity of 0kb, and sometimes not even the file. But I once replaced my SD card and run it successfully before. I don't know why.

[image]
[image]

CaptainiFalcon, 2022-04-01:
I meet the same problem as @xiaoxiaochuzm

JiaoXianjun, 2022-04-01:
Could you share us the exact information of:
1. The step by step commands/operations you have run (also tell us that operation is performed on board or on computer)
2. The output and phenomenon you observed in each step

xiaoxiaochuzm, 2022-04-02:
I'm so sorry to bother you. I think I made a low-level mistake. I didn't turn off the computer firewall in time, which led to the received message of 0.Thank you very much for your timely reply.

JiaoXianjun, 2022-04-02:
No worry. Good luck. 

Seems like you are using windows? We always use Linux.

closed 2022-04-02 by JiaoXianjun

chenmo234, 2024-05-22:
Hello, sorry to bother you. May I ask you some questions regarding IQ capture in OpenWiFi? Here is my contact information: c2385405729.