open-sdr/openwifi #152: Question for frequency offset

csy990406, 2022-04-01:
Thanks for your efforts on this great project.  I am first grade master student  and new to this device. Mainly to to identify devices by using frequency offset information. I have been using your CSI collection tool for a while but encountered a problem recently.

Like the example in the csi.md, I was running openwifi in the monitor mode and trying to capture CSI of those packets from the device with specific MAC address. I used three devices: iPhone 8,  ASUS notebook and nexus 5.  Different devices should have different frequency offsets. However the frequency offsets I capture all have a lot of the same value like 6216.99 Hz, 12433.98 Hz, 18650.97 Hz, etc. Why these points are all multiples of 6216.99 Hz? (I found that the demo you posted on Zhihu seems to be the same. https://zhuanlan.zhihu.com/p/263071541) 

Sincerely,
Siyu Chen

JiaoXianjun, 2022-04-01:
Thanks for reporting this. Yes we notice this as well. If you follow this CSI radar app note: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/radar-self-csi.md , in principle the frequency offset should be 0, because the received signal is from the same board. The Tx and Rx Lo should lock together, so there shouldn't be any frequency offset. But as you can see, the frequency offset is not 0 again.

I believe it is related to the limited bit width in the receiver implementation of frequency offset estimation and compensation .
The core component of frequency (phase rotating) algorithm is a lookup table that splits 2*pi circle/full-360-degree to many pieces, and fill the corresponding sine or atan/tan value to at the corresponding angle.

We need to check further on this to confirm.

csy990406, 2022-04-03:
Thanks for your reply. I verified the radar-self-csi.md, the result is as you said the frequency offset is not 0. Looking forward to your solution.

JiaoXianjun, 2023-01-28:
Now it might already be fixed. At least in the case of newly supported FPGA internal loopback:

https://github.com/open-sdr/openwifi/releases/tag/v1.4.0

nastyapplepie, 2023-09-22:
焦博士，我成功了完成了您的CSI信息收集的quick start，但是我不太清楚moniter模式下我收集到的信息是什么，以及通信频段是什么，收发设备是什么。能麻烦您简要解释一下吗，不胜感激。

JiaoXianjun, 2023-09-26:
> 焦博士，我成功了完成了您的CSI信息收集的quick start，但是我不太清楚moniter模式下我收集到的信息是什么，以及通信频段是什么，收发设备是什么。能麻烦您简要解释一下吗，不胜感激。

Please check this discussion:

https://github.com/open-sdr/openwifi/discussions/344

nastyapplepie, 2023-10-11:
焦博士您好，我是东南大学信息学院的大四学生，目前在关注您的openwifi项目。在按照您CSI collect的教程操作的时候，遇到了一些无法理解的困惑。如图（随附件所示），我在成功建立电脑与开发板的通讯之后开始用python脚本收集CSI信息，之后再用matlab分析收集到的信息的时候，我发现均衡器的那一张图更近似二维高斯分布而不是您在其他地方演示的16QAM星座图。目前尚未能够排查出问题所在。一切都是按照您的文档操作的，在电脑上ping板子的IP地址也可以成功。期待您的回复指导，不胜感激。
又：所使用的开发板是尼普顿的。

chenmo234, 2024-07-17:
Hello, sorry to bother you. May I ask you some questions ? I am also working on using frequency offset to distinguish devices.Here is my contact information: c2385405729.

chenmo234, 2024-07-17:
> 感谢您为这个伟大项目所做的努力。我是一年级的硕士生，是这个设备的新手。主要是通过使用频率偏移信息来识别设备。我已经使用您的 CSI 收集工具一段时间了，但最近遇到了一个问题。
> 
> 与 csi.md 中的示例一样，我在监控模式下运行 openwifi，并尝试从具有特定 MAC 地址的设备捕获这些数据包的 CSI。我用了三台设备：iPhone 8、华硕笔记本和nexus 5。不同的设备应具有不同的频率偏移。但是，我捕获的频率偏移都有很多相同的值，例如 6216.99 Hz、12433.98 Hz、18650.97 Hz 等。为什么这些点都是 6216.99 Hz 的倍数？（我发现你在知乎上发布的演示似乎是一样的。https://zhuanlan.zhihu.com/p/263071541)
> 
> 此致 Siyu Chen

Hello, sorry to bother you. May I ask you some questions ? I am also working on using frequency offset to distinguish devices.Here is my contact information: c2385405729.