open-sdr/openwifi #153: Question about AMPDU

czh869452912, 2022-04-02:
Thanks for your efforts on this great project. I am a graduate student at XJTU. Our team is looking forward to doing some tests on MPDU aggregation. From the provided documentation, OpenWiFi allows us to turn on\off the MPDU aggregation. We wonder whether we can control the number of MPDUs to aggregate and operate manually configured 802.11n package injection with custom AMPDU configuration. Thanks a lot for your help.

Sincerely,

Zuhao Chen

JiaoXianjun, 2022-04-02:
@mmehari  could you answer?

mmehari, 2022-04-02:
You can provide an upper limit to the number of aggregated packets but you can not enforce a fixed value.

For openwifi, it is set to a fixed value 4 but that can be changed. https://github.com/open-sdr/openwifi/blob/0e84893f089ee8b1c0c94eaa87118ee6c6cef03e/driver/sdr.c#L2039-L2040

Regarding AMPDU packet injection functionality, I don't think it is supported out of the box and you need to patch the kernel for such functionality.

The linux kernel handles packet aggregation by setting the **IEEE80211_TX_CTL_AMPDU** information flag into the sk_buff and that is what we check in the driver if the packet is to be aggregated or not.https://github.com/open-sdr/openwifi/blob/0e84893f089ee8b1c0c94eaa87118ee6c6cef03e/driver/sdr.c#L1078

In this regard if aggregation is supported, the **IEEE80211_TX_CTL_AMPDU** information flag will be set in the injector program (inject_80211.c) and parsed by the **ieee80211_parse_tx_radiotap** kernel function (https://elixir.bootlin.com/linux/v4.14/source/net/mac80211/tx.c#L1924). As far as I am aware, there is not support for this and you have to patch it yourself.

One possibility could be setting the **IEEE80211_TX_CTL_AMPDU**  information flag here (https://elixir.bootlin.com/linux/v4.14/source/net/mac80211/tx.c#L1945).

Hope you find this helpful

JiaoXianjun, 2022-04-02:
Another possibility could be that override/set the aggregation flag in driver to let the fpga do (possible <=4) aggregation on the packet. This indication could be passed via some unused field in radiotap of the injected packet (by packet injection program)

czh869452912, 2022-04-06:
Thanks a lot for your suggestion! We are going to try both attempts to figure out how to achieve it.

closed 2022-04-27 by JiaoXianjun