open-sdr/openwifi #154: When I connected to AP, I encountered the following error

Mrxia111, 2022-04-03:
[image]
As shown in the figure, when I want to connect to AP, the following error occurs
[image]
this is my wpa-connect.conf file.
[image]
how can i address? could give me some advice.

JiaoXianjun, 2022-04-03:
Could you answer questions in our template? Without knowing those info, it is difficult for us to tell what is the issue.

JiaoXianjun, 2022-04-03:
can you use other client (phone/laptop) connect that AP?

what is the channel number of that AP?

can you set your AP to a channel in 5GHz instead of 2.4GHz?

what is the distance from your AP to openwifi. 

Can you configure your AP into 11ag or 11n mode (not ac or ax).

Can you run “ iwlist sdr0 scan” on openwifi board, and share me output.

JiaoXianjun, 2022-04-03:
Also check your antenna connection. You need two antennas connected to tx and rx port. What is your RF board?

JiaoXianjun, 2022-04-03:
If you follow the quick start guide to setup openwifi AP, can you use your phone/laptop to connect to openwifi AP?

Mrxia111, 2022-04-03:
many computer can use this AP 

this is picture which is using my phone to connect AP 

that AP is close

I can run “ iwlist sdr0 scan” on openwifi board,

closed 2022-04-13 by JiaoXianjun