open-sdr/openwifi #155: In order to change SAMPLE_RATE from 20 to 40MHz what items would need to be changed in the database ?

chs9996, 2022-04-08:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

JiaoXianjun, 2022-04-08:
Assume you want a non-standard 40MHz wifi. Assume we do it in quick and dirty way. Not decent. Simply scale the current 20MHz openwifi to 40MHz.

A decent way should be using higher sampling rate (>=40MHz) for ad9361, and using a channel FIR filter to have better channelization. But let's assume we use 40MHz sampling rate directly from ad9361.

change the bw (in_voltage_rf_bandwidth/out_voltage_rf_bandwidth) in rf_init_11n.sh to >= 40MHz

disable the FIR (our FIR is for 20MHz) in rf_init_11n.sh via in_voltage_filter_fir_en/out_voltage_filter_fir_en

remove the decimation operation 40M-->20M in openwifi-hw/ip/rx_intf/src/adc_intf.v. bypass that decimation to make sure you can have 40M samples through

remove the interpolation (zero insertion) 20M-->40M in openwifi-hw/ip/tx_intf/src/dac_intf.v. (see line of: wren_internal_reg <= (counter==0 || counter==1); // 0 for IQ from bb fifo, 1 for zero insertion)

the phy tx and rx core might need some processing time for each sample, so to handle the higher sampling rate, you might want to use higher clock rate for FPGA. Before build FPGA, you might want to change NUM_CLK_PER_US from 100 to 200 at the beginning of openwifi.tcl. example: openwifi-hw/boards/adrv9361z7035/openwifi.tcl

after the FPGA is built, you can try. depends on how lucky are you. above is just my quick screening because we never think and try.

all the rssi stuffs are calibrated in our 20MHz ad9361 and FPGA setting. it is related to CCA (Channel sensing) accuracy, receiver sensitivity setting, etc. so you really need to have good understanding of whole wifi chip and handling all the details (not limited to above).

good luck.

chs9996, 2022-05-06:
The adc clk is 40mhz anyway which is what i needed, so have what i need. This is really good information above. Thank you

closed 2022-05-06 by chs9996