open-sdr/openwifi #156: Question：

sada45, 2022-04-09:
0. Could you send email to xianjun.jiao@ugent.be to introduce your self?

1. Our image is used directly or you build your own image?

2. What is your own modification?

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision

4. Board/hardware type

5. WiFi channel number

6. Steps to reproduce the issue, and the related error message, screenshot, etc

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"

8. Describe your debug efforts by: https://github.com/open-sdr/openwifi/blob/master/doc/README.md#Debug-methods

9. Any other thing we need to know for helping you better?

JiaoXianjun, 2022-04-09:
We only developed the driver and FPGA. For the Linux kernel and mac80211 framework above, it is what it is. 

So I don’t know. 

You need to find out by yourself. Looking forward to your answer.

sada45, 2022-04-09:
Alright, we have just order a antSDR to try this project :)

closed 2022-04-13 by JiaoXianjun