open-sdr/openwifi #158: Issue about wifi signal strength.

852432070, 2022-04-12:
Thanks for your efforts to build this open source wifi project! I use a zed_fmcs2 device with openwifi in my office, and I want to capture some packages from openwifi. However, the wifi signal is too weak that I can't get a good wifi connection in the office environment. Is there any way to make it stronger just using the existing software?(Maybe some scripts to add a gain to the rf module.)

JiaoXianjun, 2022-04-13:
You can try to enlarge the value of register 13 of tx_intf (Check https://github.com/open-sdr/openwifi/blob/master/doc/README.md)
By default it is 250 for good EVM and making sure long packet of all MCSs get through. If you increase it, EVM could be bad, and some long packet of some MCS won't get through. But you can try.

JiaoXianjun, 2022-04-13:
Read this app note: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#tx-power-config

852432070, 2022-04-14:
Thanks a lot, that really helps!

closed 2022-04-14 by JiaoXianjun