open-sdr/openwifi #159: Failed to capture the injected packages of openwifi.

852432070, 2022-04-14:
I am a student who want to inject wifi package using openwifi. I am using zed_fmcs2 as openwifi device and 9300 nic running wireshark to capture wifi packages. According to before tests, I can capture packages from openwifi in channel 44 when in AP mode.
This is part of the injected package configuration:
```
/* IEEE80211 header */
static u8 ieee_hdr_data[] =
{
	0x08, 0x02, 0x00, 0x00,             // FC 0x0801. 0--subtype; 8--type&version; 02--toDS0 fromDS1 (data packet from DS to STA)
	0x66, 0x55, 0x44, 0x33, 0x22, 0x11, // BSSID/MAC of AP
	0x66, 0x55, 0x44, 0x33, 0x22, 0x22, // Source address (STA)
	0xab, 0xab, 0xab, 0xab, 0xab, 0xab, // Destination address (another STA under the same AP)
	0x10, 0x86,                         // 0--fragment number; 0x861=2145--sequence number
};
```
The commands I used are as follow:
```
./wgd.sh 
./monitor_ch.sh sdr0 44
./inject_80211/inject_80211 -m n -r 0 -n 9999 -s 24 sdr0
```
In ideal conditions, I will capture packages with the filter `wlan.addr==66:55:44:33:22:11`, while no packages with this mac address is received in channel 44.
And I got these logs:
```
sdr,sdr openwifi_tx: WARNING gen_ht_duration_id wrong pkt type!
sdr,sdr openwifi_tx: WARNING packet is not QoS packet!
```
Then I changed some regs as the issue `Packet injection function cannot send data packets continuously #117` mentioned:
```
./sdrctl dev sdr0 set reg xpu 11 16
./sdrctl dev sdr0 set reg xpu 1 4294967296
./sdrctl dev sdr0 set reg drv_tx 7 7
```
And I used this command to inject packages:
```
./inject_80211/inject_80211 -m n -r 0 -n 10 -s 64 -t d -e 8  sdr0
```
Still no package captured...
And the log is as follows:
```
sdr,sdr openwifi_tx: 92B RC8 130M FC0288 DI003c ADDR665544332201/665544332202/abababababab flag400100d9 QoS91 SC2145_0 retr0 ack1 prio3 q3 wr19 rd0
```
Is there any problem in my previous operations? If not, what could cause the situation that no injected package can be received?

Thanks for your efforts to develop openwifi platform!

JiaoXianjun, 2022-04-14:
The filter (the filter wlan.addr==66:55:44:33:22:11) might be the issue. Please debug without the filter.

852432070, 2022-04-16:
OK, I'll try this. Thanks!