open-sdr/openwifi #160: Can't recv beacon frame and recv a bad probe response

kong1843, 2022-04-15:
I appreciate your work very much. It's excellent.but I have some problems.

My openwifi device is adrv9364z7020,the openwifi version is v1.3.0.
I set the sdr0 monitor mode by './monitor_ch.sh sdr0 6',but the pcap of tcpdump doesn't include beacon frame.

I used python code to send 802.11 frames and recv 802.11 frames in monitor mode.This can work.
'''
    ETH_P_ALL = 3
    s = socket.socket(socket.AF_PACKET, socket.SOCK_RAW, socket.htons(ETH_P_ALL))
    s.settimeout(5)
    s.bind(('sdr0', ETH_P_ALL))
    s.send(pkt）
    pkt = s.recv(1024)
'''
I can send beacon frame\probe request\auth and the others,I also can recv probe response\auth frame and the others,but no beacon frame.

Another question.
I can't recv the 802.11 frame with destination MAC address is sdr0's.
for example,I send a probe request frame(source mac addr is '66:55:44:33:22:92') by openwifi's sdr0,I only recv a malformed frame,Its binary form is as follows，
’00001a002f4800007fea970000000000100c8509c000cd010000 d400 fe00 665544332292 2acd27fb‘，

I confirm that the peer AP device has sent a normal probe response frame, and I can also receive the probe response  with the  destination mac address of other devices.

Have you ever encountered this problem? Do you have any suggestions？Thanks very much.

JiaoXianjun, 2022-04-16:
Not understand well of the issue. Please describe step by step command/operation for reproducing the issue.

kong1843, 2022-04-18:
cd ~/openwifi/
./wgd.sh
./monitor_ch.sh sdr0 6
tcpdump -i sdr0 -w out.pcap
I see the packets with wireshark,but most of them are incorrect packets, except probe request \ response packets. 

cd ~/openwifi/
./wgd.sh
./monitor_ch.sh sdr0 6
python recv_frame.py
I run the recv_frame.py，print the packets received by sdr0. The same was observed

out.pcap and recv_frame.py are attached
[files.zip](https://github.com/open-sdr/openwifi/files/8504022/files.zip)
.

JiaoXianjun, 2022-04-22:
Hello,

If I run tcpdump -i sdr0 | grep Beacon, I can see lots of beacon.

Current openwifi works only for 20MHz OFDM 11a/g/n mode, so you need to make sure your target devices (AP, Client) not using 40MHz/80MHz/160MHz or 11b/ac/ax mode.

incorrect packets could be due to weak signal or interference. You can let the receiver only works for strong signals. Check this app note: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/frequent_trick.md#receiver-sensitivity-control

closed 2022-05-10 by kong1843