open-sdr/openwifi #161: Does openwifi support stand-alone operation?

gxhen, 2022-04-21:
Hi Dr Jiao,

Could you please tell me whether Openwifi supports stand-alone operation? That is, I would like to process CSI using the ARM core onboard instead of transmitting them to a PC with an Ethernet cable. Is it theoretically possible and do I need to heavily modify the driver?

Thanks in advance.

JiaoXianjun, 2022-04-21:
of course openwifi can run stand-alone, just like any other wifi rounter (they don't need screen and pc, they run with power supply and ethernet). openwifi is a full linux computer!

if you read the CSI app note carefully: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md
you will realize that: the CSI information is collected on board by the userspace program [side_ch_ctl.c](https://github.com/open-sdr/openwifi/blob/master/user_space/side_ch_ctl_src/side_ch_ctl.c), and then relay to the external PC.

So, if you want to process CSI on board, you need to read that side_ch_ctl.c and modify it according to your needs to make all necessary processing on board (maybe you can put your processing into side_ch_ctl.c, and remove the CSI relay via UDP to PC).

gxhen, 2022-04-21:
Very clear instruction and it sounds indeed doable. Thanks very much for this excellent project.

closed 2022-04-22 by JiaoXianjun