open-sdr/openwifi #162: Problem with axi_ad9361 module in vivado

SHUMT18, 2022-04-24:
The module of axi_ad9361 has four channels of enable and valid signal output. What is the function of these signals? If I set all these signals to 1, can the module work normally?

JiaoXianjun, 2022-04-24:
Sorry that module is not designed by us. The correct place to ask question about that module is analog devices tech forum. 
https://ez.analog.com/

SHUMT18, 2022-04-24:
If I set both adc_valid and dac_valid in the openwifi module to 1, can the module work normally?

JiaoXianjun, 2022-04-24:
Just try it!

JiaoXianjun, 2022-04-24:
In the case of openwifi, they are always 1

SHUMT18, 2022-04-24:
I encountered a problem in the project, what is the function of the gpio pin of openwifi. Is it related to the control output of ad9361?

JiaoXianjun, 2022-04-24:
Can you specify which gpio signal are you talking? (Which port of which FPGA module/file)

SHUMT18, 2022-04-24:
openwifi ip - gpio_status

SHUMT18, 2022-04-25:
I found a few modules in the bd diagram of your vivado project. I don't quite understand their function. Could you please give me an answer?
Here are the names of these IPs:
axi_gregs
axi_pz_xcvrlb
axi_icc_main

JiaoXianjun, 2022-04-26:
> openwifi ip - gpio_status

In the driver (sdr.c), we use ad9361_ctrl_outs_setup to set the gpio output of ad9361 to represent the real-time Rx gain. You can check the manual of ad9361 for more info.

JiaoXianjun, 2022-04-26:
> I found a few modules in the bd diagram of your vivado project. I don't quite understand their function. Could you please give me an answer? Here are the names of these IPs: axi_gregs axi_pz_xcvrlb axi_icc_main

These modules are not written by us, so we do not know.

closed 2022-04-26 by JiaoXianjun

SHUMT18, 2022-04-26:
> > openwifi ip - gpio_status
> 
> In the driver (sdr.c), we use ad9361_ctrl_outs_setup to set the gpio output of ad9361 to represent the real-time Rx gain. You can check the manual of ad9361 for more info.
I have found it Thanks

SHUMT18, 2022-04-26:
> > I found a few modules in the bd diagram of your vivado project. I don't quite understand their function. Could you please give me an answer? Here are the names of these IPs: axi_gregs axi_pz_xcvrlb axi_icc_main
> 
> These modules are not written by us, so we do not know.

Are these IPs not used in the openwifi project? Because I didn't find the interaction of these IPs with openwifi.
I think the openwifi project may be made on the basis of some projects, whether these ips are the remnants of these projects ,these IPs has no effect on openwifi

JiaoXianjun, 2022-04-26:
As we said here, openwifi is based on ADI HDL reference design. On top of that (including the modules you asked), we add openwifi ip: https://github.com/open-sdr/openwifi#porting-guide

SHUMT18, 2022-04-27:
When I synthesize the openofdm_rx project, the following error will be reported
[Synth 8-5796] RAM (sync_short_inst/mag_sq_avg_insti_0) has conflicting writes using multiple ports with same address
I didn't find a solution for this error on the web
Did you also encounter this problem when you were working on a project?

SHUMT18, 2022-04-28:
OK Thanks

SHUMT18, 2022-04-29:
I would like to ask you a question about the device tree. How did you generate your device tree?
I see that you downloaded an img file from the official website of adi and made it into a Linux system. This system includes the device tree of adi, so how do you add your own devices such as sdr xpu to the device tree?

JiaoXianjun, 2022-04-29:
https://github.com/open-sdr/openwifi#porting-guide

SHUMT18, 2022-05-01:
I understand that you got the dtb from the dts file, so how did you put the modified dts file into the system?
Convert dts to dtb through dtc tool?

SHUMT18, 2022-05-13:
I found that you updated the project two months ago, you gave the clock given from ad9361 to the bus clock of openwifi, but I did not see that you modified it in the device tree, the bus clock of openwifi in the device tree is still given by ARM. Does this affect engineering?
I am using board adrv9361z7035

JiaoXianjun, 2022-05-13:
It doesn’t matter.

SHUMT18, 2022-05-19:
A network card named wlan0 appeared on the board after I loaded sdr.ko, After I load hostapd-openwifi.conf, I can search for the network openwifi on my phone, but I can't connect to it, it shows error on putty:
"xilinx-vdma 80410000.dma: Channel ee43d810 has errors 10, cdr 3f048000 tdr 3f044000"
This is the dma connected to the rx_intf module. I don't know if this error has any effect. I speculate that the AP can send broadcast signals at present, but the information sent by ue cannot be received by the AP.