open-sdr/openwifi #164: Problem of gpio in ADRV9361-Z7035 project

SHUMT18, 2022-04-26:
In the top of the vivado project of ADRV9361-Z7035, there are many gpio interfaces such as gpio_bd, gp_out, gp_in, etc. I found out that these gpio ports are connected to the daughter board (as you said we can use ADRV1CRR-FMC or ADRV1CRR-BOB) I would like to know if these ports connected to the daughter board have any effect on the implementation of openwifi. Because my board is not an official board, he does not have these interfaces

JiaoXianjun, 2022-04-26:
Sorry that those gpio interfaces are not created by us. You need to track/understand them by yourself.

closed 2022-04-26 by JiaoXianjun

SHUMT18, 2022-04-26:
> Sorry that those gpio interfaces are not created by us. You need to track/understand them by yourself.

ok Thanks