open-sdr/openwifi #169: How can I get timestamp based on TSF when I use tcpdump to catch the packets on board?

GuanLiliang, 2022-05-11:
When I use tcpdump, I can get timestamp like this, 
[image]
Is this based on TSF? I don't think it is. Maybe it is based on starting up time.
If not, how can I get it based on TSF?

gxhen, 2022-05-11:
u can transfer the captured .pacp file from the board to your PC via SSH. Then use Wireshark to analyze it. TSF timestamp is in the 802.11 radio information.

CaptainiFalcon, 2022-05-31:
I am learning 802.11* protocal recently. I think the things about time synchronization is interesting.

When I followed the instructions  in [csi.md](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md) to capture CSI and ran “test_side_info_file_display.m”, I got the timestamps as shown below.

[image]

The unit of it's value is us, （18,000,000 us = 18 second;  28,000,000 us =28 second）so it seem like that the timestamp in CSI (side_info.txt) is timeing from when I setup the 'sdr0' in board by running ‘wgd.h' (but not from when the ubuntu system boot in board).

After a while, I captured CSI again, and got :

[image]
(1600000000 us=26.6666667 minute)

And I saw [timestamp in packet sniffer ](https://github.com/open-sdr/openwifi/issues/139#issuecomment-1051988468) is set by 'openwifi_rx_interrupt()', so I think the timestamp in packet sniffer is the time of Untuntu system (in board) that can be got by running 'date' in the shell of ubuntu in board and it's something like '2022-01-25 00:20:29'(the date in my system is that). 

Actually, the timestamp in tcpdump run in my board is "2022-01-25 **:**:**), same as what I thought. 

In short, I don't understand "If you want to relate the CSI information to the WiFi packet, you need to capture WiFi packets (tcpdump/wireshark/etc) while capturing CSI. Then you can match the timestamp (TSF timer value) between WiFi packet and CSI information, because this is the unique same identity of a Wifi packet and related CSI information." in [csi.md](https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/csi.md).

Specifically, I don't understand what is "timestamp (TSF timer value) betweem WiFi packet and CSI". The differnece of timestamp in CSI and packet sniffer is beyond my knowledge.

And I also want to know how to align the time in CSI and packet sniffer.

Apologies for asking some basic question and the English is not standard.

I am looking forward for your reply, thank you very much.

JiaoXianjun, 2022-05-31:
You have timestamp (in us) on the csi data. You also have timestamp (in us) on the wifi packet captured by packet sniffer (as in the screenshot at the beginning of this issue). They are in the same timeline (both are based on tsf timer). So you can know the relation between csi data and the packet.

CaptainiFalcon, 2022-05-31:
Understood.

Now, the problem change to "align ONE CSI to the corresponding packet".  I think it is difficult and hava no idea about that when do not modify the hardware design of openwifi. 

Can you show me the way? 

Thank you very much again.

JiaoXianjun, 2022-05-31:
You have two data sets of time stamp, one from csi, one from packet. Just do what ever you want!

CaptainiFalcon, 2022-05-31:
Ohh, thanks a lot！

CaptainiFalcon, 2022-06-03:
As shown in the figure below:

[image]

Why do two or more csi have the same timestamp ?

closed 2022-06-18 by JiaoXianjun