open-sdr/openwifi #171: Port Openwifi to my own board

cjhonlyone, 2022-05-17:
Hello, Jiao
Sorry to bother you. I port open wifi to my own board but failed. Console showed AP has initialized successfully, but I didn't find openwifi ssid on my phone.

1. Our image is used directly or you build your own image?
Build my own image
I have built two images, one for Zedboard and the other for my ownboard.
Zedboard works normally, but my board can not transimit Wifi signal.

2. What is your own modification?
Differences between Zedboard and my own board:
     zedboard    my
1. xc7z020            xc7z030
2. iic_fmc         connected to unused pins
3.  ad_iobuf #(.DATA_WIDTH(49)) i_iobuf_gpio (
    .dio_t ({gpio_t[50:49], gpio_t[46:0]}),
    .dio_i ({gpio_o[50:49], gpio_o[46:0]}),
    .dio_o ({gpio_i[50:49], gpio_i[46:0]}),
    .dio_p ({ gpio_muxout_tx,
              gpio_muxout_rx,
              gpio_resetb,
              gpio_sync,
              gpio_en_agc,
              gpio_ctl,
              gpio_status_dummy,
              gpio_bd}));

  ad_iobuf #(.DATA_WIDTH(3)) i_iobuf_gpio (
    .dio_t ({gpio_t[46:44]}),
    .dio_i ({gpio_o[46:44]}),
    .dio_o ({gpio_i[46:44]}),
    .dio_p ({ gpio_resetb,
              gpio_sync,
              gpio_en_agc}));

3. Versions: OS, Vivado, openwifi/openwifi-hw repo branch and commit revision
openwifi master
openwifi-hw master

7. Describe your debug efforts by Linux native tools, such as tcpdump and "cat /proc/interrupts"
I added ila to capture dac_data_0/1/2/3 and found they stay all zero all the time.
To address the problem, I wrote arbitrary IQ to tx_intf and successfully triggered my ila.

Then I viewed dmesg of my board, I noticed there is no print of "openwifi_tx_interrupt".
I think the problem maybe is about dma transfer?
Can you give me some advices to debug this?
Thank you!

This is my dmesg log
[z7030log.txt](https://github.com/open-sdr/openwifi/files/8704860/z7030log.txt)
.

YideTian, 2022-05-17:
@cjhonlyone hi! my board is also based on xc7z030.I changed the constain and build the boot bin by instruction. But i still can't see any output from my uart console.Could you tell me which part of the design should be modified to fit the xc7z030? thank you!

cjhonlyone, 2022-05-17:
> 

Hi, I think you should check these things.
1. Select right DDR Part
2. Select right uart in bootargs
3. Comment some device not in your board in dts
But first step is to see the output of u-boot.
Good Luck!

cjhonlyone, 2022-05-23:
I found the point!
Openwifi read gpio_status to calculate RSSI. I disconnect the port, so as to tx_intf can not send any packet for this channel is already busy.
But when I connect the gpio_status port to physical pin, RSSI is still wrong.
So I change the magic -78dBm in sdr.c to 0dBm, that works.
I will find out the reason why RSSI isn't right in few days.

closed 2022-06-07 by cjhonlyone