open-sdr/openwifi #173: some question about linux runing on my own board

YideTian, 2022-05-17:
hello, My own board is based on xc7z030 , and i have done some modification for my board:
1. Modify the DDR configuration and MIO constrain. The same constrain is been tested for another SD card, which  has petalinux inside. The console shows that the system is boot successfuly. So the uart(UART0) chain and DDR would be right.
2. I modified the .xdc file and change the pin location to my own board.
3. Add two led module to system_top. one module is driven by PL side clock, and another clock is from zynq ip FCLK_CLK0. 
4. I'm using "adrv9361z7035" as my board name to generate boot bin.
then i "Build openwifi Linux img from scratch" as README construction said.
my BOOT partion is like this:
[image]
the problem is:
Those two led is flashing on power up. so i guess fsbl is success. But i haven't seen any output in uart console yet. Could you help me find the problem? Thank you!

cjhonlyone, 2022-05-17:
Hi, YideTian
You should check hdf status and u-boot compiling process.
u-boot is not too much association with hardware.
Check whether FPGA is programed properly by fsbl.
Good Luck!

YideTian, 2022-05-20:
@cjhonlyone Hi, Thank you for your response. 
I have checked the DDR, I think it's ok ,The hdf is up to date. Then i modify the device tree and run update_sdcard.sh. The result has no difference. I guess u-boot should be changed properly.
But i don't know how to deal with u-boot. May i ask for your email address? So that i can ask you about some details. Thanks again for your kindness help!

cjhonlyone, 2022-05-21:
@YideTian cjh416593819@gmail.com

YideTian, 2022-05-23:
@cjhonlyone Thank you!

closed 2022-06-18 by JiaoXianjun