open-sdr/openwifi #177: Questions about synchronization

BZZDKBRS, 2022-06-01:
Does openwifi realize the synchronization function? If so, how is it achieved? thank you!

JiaoXianjun, 2022-06-01:
I don't know. Could you explain in detail what is the synchronization you want?

JiaoXianjun, 2022-06-01:
The driver and FPGA only passes the necessary information between upper layer/Linux and the antenna (such as the packet and its TSF timestamp, etc.), and it only executes setting from upper layer/Linux.

The functionality you described is in upper layer I think.

Or, you have all driver and Linux source code, I think you can also try to implement this functionality.

JiaoXianjun, 2022-06-09:
You can also check our paper on synchronization on this page: https://github.com/open-sdr/openwifi/blob/master/doc/publications.md

BZZDKBRS, 2022-06-09:
Thank you very much.

closed 2022-06-18 by JiaoXianjun