open-sdr/openwifi #179: Questions about ad-hoc network

lydenlau, 2022-06-10:
Hello，now we  want to add batman-adv to openwifi，How should we do it？

JiaoXianjun, 2022-06-10:
1. I don't know what is batman-adv, so I don't know.
2. We do support ad-hoc mode: https://github.com/open-sdr/openwifi/blob/master/doc/app_notes/ad-hoc-two-sdr.md
3. If I assume batman-adv is like a Linux software/kernel-module, in principle it could run with openwifi. Why?
4. Because openwifi offers two things: the WiFi chip/card (in FPGA), and its Linux driver. The driver is Linux mac80211 framework compatible. So, from Linux point of view, openwifi chip/card is yet another WiFi chip/card, just like any other COTS WiFi chip/card.
5. On board, we also installed a Linux image. The current version is Linaro, the next version will be Raspbian, the same as Raspberry PI.
6. I don't know what is the current Linux environment for batman-adv. You can either try to install batman-adv into the openwifi onboard Linux, or try to install the Linux you need for batman-adv onto the openwifi board (of course you need to pick our FPGA image and driver into your Linux to run openwifi).

lydenlau, 2022-06-13:
Thanks a lot, I'll try your advice.

lydenlau, 2022-06-27:
Sorry for the break，there is another question. I'm trying  to add the batman-adv.ko to the openwifi linux.My compilation steps are as follows：
1.Building the Zynq Linux kernel and devicetrees from source，the source link is https://github.com/analogdevicesinc/linux/tree/b6e379910a11af77e6500ed8b0895006e471a279
(I uesd the Xilinx toolchain vivado 2019.1 at ubuntu 20.04)
2. Modified makefile（EXTRAVERSION = -gb6e379910a11） and .config（CONFIG_LOCALVERSION="-dirty"） to make sure the vermagic is 4.14.0-gb6e379910a11-dirty( the same as the linux) , then  I successfully compiled batman-adv.ko
3. when I insmod batman-adv.ko, it show:
root@analog:~/openwifi# insmod batman-adv.ko
insmod: ERROR: could not insert module batman-adv.ko: Invalid module format

We look forward to hearing from you, I want to know how you guys compile the module and run it, just like the openofdm_tx.ko and other modules.

Additional information about the module is as follows：
root@analog:~/openwifi# modinfo batman-adv.ko
filename:       /root/openwifi/batman-adv.ko
alias:          net-pf-16-proto-16-family-batadv
alias:          rtnl-link-batadv
version:        2017.3
description:    B.A.T.M.A.N. advanced
author:         Marek Lindner <mareklindner@neomailbox.ch>, Simon Wunderlich <sw@simonwunderlich.de>
license:        GPL
srcversion:     78D5FE0D577C2CC76BC6FFC
depends:
name:           batman_adv
vermagic:       4.14.0-gb6e379910a11-dirty SMP preempt mod_unload modversions ARMv7 p2v8

lydenlau, 2022-06-28:
> Seems your Vivado/Xilinx toolchain version and kernel config are different from ours. This might cause the issue. To check how do we create Linux kernel with our own config, see:
> 
> https://github.com/open-sdr/openwifi#build-openwifi-linux-img-from-scratch
> 
> https://github.com/open-sdr/openwifi/blob/master/user_space/update_sdcard.sh
> 
> https://github.com/open-sdr/openwifi/blob/master/user_space/prepare_kernel.sh

Thanks a lot！I'm trying change my vivado to 2018.3，Is there anything else to be aware of?