open-sdr/openwifi #180: 修改发射频率为1.4G 的问题

reuncc, 2022-06-18:
我使用 仓库:https://github.com/MicroPhase/openwifi 的镜像用于  MicroPhase E310 (antsdr) 当前仓库中我看到 可以通过sdrctl 命令设置发射频率, 但是在我所使用的仓库中发现   ./sdrctl dev sdr0 set reg rf 1 3500  指令无效,所有 reg fr 指令设置之后,值始终 fffffff

我尝试直接下载 https://github.com/open-sdr/openwifi 中的 antsdr 的镜像, 按Quick start 的步骤操作, 不进行任何其他操作 ，
执行./wgd.sh 报错,
执行./fosdem.sh 报错,
错误是 是接口驱动找不到, 
因为我刚刚接触ANTSDR 所以自己很难找到真实的问题所在,
===========
我想请问一下, 我的设备{antsdr} 在下载在镜像 [antsdr	MicroPhase enhanced ADALM-PLUTO Notes	Done	32bit img	NO need]
并复制对应的 uImage  BOOT.bin 之后  还需要进行其他什么操作 才能正常运行呢

=========
期待您的回答！

JiaoXianjun, 2022-06-19:
1. Please use English to raise questions.
2. You need to tell me the detailed/exact each step operation (commands, output, screenshot, etc) you have done while you following the quick start guide: https://github.com/open-sdr/openwifi#quick-start .

reuncc, 2022-06-20:
1.Download Image: https://users.ugent.be/~xjiao/openwifi-1.3.1-wilsele-32bit.img.xz(Source:https://github.com/open-sdr/openwifi)
2.Write Image to SDCard
3.Copy files
	BOOT: /openwifi/antsdr/BOOT.BIN >> BOOT:/
	BOOT: /openwifi/antsdr/devicetree.dtb >> BOOT:/devicetree.dtb
	BOOT: /openwifi/zynq-common/uImage >> BOOT:/uImage
4.Connect two antennas to RXA/TXA ports.
5.PC commands:
     	ifconfig eth0  192.168.10.123 
	ssh root@192.168.10.122
6.Execute command in'root@192.168.10.122':

root@analog:~# cd openwifi
root@analog:~/openwifi# ./wgd.sh
===================./wgd.sh====Print Begin=================
usage:
  Script for load (or download+load) different driver and FPGA img without rebooting
  no  argument: Load .ko driver files and FPGA img (if system_top.bit.bin exist) in current dir with test_mode=0.
  1st argument: If it is a NUMBER, it will be assigned to test_mode. Then load everything from current dir.
  1st argument: If it is a string called "remote", it will download driver/FPGA and load everything.
  - 2nd argument (if exist) is the target directory name for downloading and reloading
  - 3rd argument (if exist) is the value for test_mode
  1st argument: neither NUMBER nor "remote" nor a .tar.gz file, it is regarded as a directory and load everything from it.
  - 2nd argument (if exist) is the value for test_mode
  1st argument: a .tar.gz file, it will be unpacked then load from that unpacked directory
  - 2nd argument (if exist) is the value for test_mode

TARGET_DIR ./
DOWNLOAD_FLAG 0
test_mode 0
tx_offset_tuning_enable 0
$TARGET_DIR is found!

network-manager stop/waiting
rmmod: ERROR: Module sdr is not currently loaded
rmmod: ERROR: Module ad9361_drv is not currently loaded
+ insmod .//ad9361_drv.ko
ad9361_drv is loaded!
.//system_top.bit.bin not found. Skip reloading FPGA.
xlnx,zynq-7000
fjdo349ujtrueugjhj
fjdo349ujtrueugjhj
+ ifconfig sdr0 down
sdr0: ERROR while getting interface flags: No such device
+ rmmod sdr
rmmod: ERROR: Module sdr is not currently loaded
+ rmmod openofdm_rx
rmmod: ERROR: Module openofdm_rx is not currently loaded
+ rmmod openofdm_tx
rmmod: ERROR: Module openofdm_tx is not currently loaded
+ rmmod rx_intf
rmmod: ERROR: Module rx_intf is not currently loaded
+ rmmod tx_intf
rmmod: ERROR: Module tx_intf is not currently loaded
+ rmmod xpu
rmmod: ERROR: Module xpu is not currently loaded
+ sleep 1
+ '[' -f fjdo349ujtrueugjhj ']'
+ '[' xlnx,zynq-7000 '!=' xlnx,zynq-7000 ']'
+ SPI_DEVNAME=spi0.0
+ DDS_DEVNAME=79024000.cf-ad9361-dds-core-lpc
+ ADC_DEVNAME=79020000.cf-ad9361-lpc
+ '[' '!' -d /sys/bus/spi/drivers/ad9361/spi0.0 ']'
+ cd /sys/bus/spi/drivers/ad9361/
+ echo spi0.0
+ echo spi0.0
+ '[' '!' -d /sys/bus/platform/drivers/cf_axi_dds/79024000.cf-ad9361-dds-core-lpc ']'
+ cd /sys/bus/platform/drivers/cf_axi_dds/
+ echo 79024000.cf-ad9361-dds-core-lpc
+ echo 79024000.cf-ad9361-dds-core-lpc
[... 83 lines cut ...]
 for additional help text.

For more details see ps(1).
+ test -f /sys/kernel/debug/iio/iio:device0/direct_reg_access
+ test -f /sys/kernel/debug/iio/iio:device1/direct_reg_access
+ device_path=/sys/kernel/debug/iio/iio:device1/
+ set +x
the end

===================./wgd.sh====Print End=================
 
root@analog:~/openwifi# ./fosdem.sh

===================./fosdem.sh====Print Begin=================
hostapd: no process found
network-manager stop/waiting
SIOCSIFADDR: No such device
sdr0: ERROR while getting interface flags: No such device
stop: Unknown instance:
isc-dhcp-server start/running, process 2098
Configuration file: hostapd-openwifi.conf
nl80211: 'nl80211' generic netlink not found
Failed to initialize driver 'nl80211'
hostapd_free_hapd_data: Interface sdr0 wasn't started
===================./fosdem.sh====Print End=================