open-sdr/openwifi #186: Does openwifi support mesh (open802.11s)?

yohanes-erwin, 2022-07-01:
Hi, I'm a researcher in FPGA design, but a beginner in Linux kernel. Currently, I'm looking for a suitable method for wireless mesh communication (MANET) for our project. I have designed PHY layer based on 802.16 on FPGA, but have difficulty in the MAC layer. Then, I found out that there is a protocol that already exists on mac802.11 which is open802.11s (mesh).

I've tried the open802.11s (it is part of the mac80211) on the openwifi design (using openwifi PHY) on the ADRV9361-Z7035 board. I ran the wgd.sh script without A-MPDU. But I didn't run the fosdem.sh script because I thought it was for the AP. I tried channels 1 and 11. I ran some linux commands like in this documentation (https://github.com/o11s/open80211s/wiki/HOWTO) to enable the mesh interface.
When I try to connect to the mesh network I get an error like this in the kernel message:
tsf : max clockdrift=0; too small to adjust
I think it's related to the timestamp, but I haven't been able to figure out what the problem is.

JiaoXianjun, 2022-07-06:
I think the error printing is not from openwifi side.
You need to find out the printing message location in the kernel (or other module) code.

yohanes-erwin, 2022-07-11:
Yes, the error printing is not from openwifi side, but from mac80211, from function mesh_sync_offset_adjust_tsf() in file mesh_sync.c.
Then, I found that the function openwifi_beacon_work() in the sdr.c calls this:
  (sdr.c) openwifi_beacon_work() ->
  (mac80211.h) ieee80211_beacon_get() ->
  (tx.c) ieee80211_beacon_get_tim() ->
  (tx.c) __ieee80211_beacon_get() ->
  (mesh_sync.c) adjust_tsf() ->
  (mesh_sync.c) mesh_sync_offset_adjust_tsf() -> msync_dbg(sdata, "TSF : max clockdrift=%lld; too small to adjust\n", ...

JiaoXianjun, 2022-07-12:
"clock drift is too small to adjust"
To my understanding, the clock drift is expected to be small, right? So this "good thing" will cause problem?

yohanes-erwin, 2022-07-13:
Yes, I think so. The smaller the clock drift the better.
But what confuses me is that this error appears every 1 second or so (in kernel message), and I also couldn't ping the other mesh client. So, I assume this error that causes the problem of not being able to ping to the other mesh client.

This is the command that I run to set the mesh network:
Board FPGA 1 (ADRV9361Z7035):
```
./wgd.sh
iw dev sdr0 interface add mesh0 type mp
ifconfig mesh0 192.168.3.91
iw dev mesh0 mesh join my_mesh
ping 192.168.3.92
```
Board FPGA 2 (ADRV9361Z7035):
```
./wgd.sh
iw dev sdr0 interface add mesh0 type mp
ifconfig mesh0 192.168.3.91
iw dev mesh0 mesh join my_mesh
ping 192.168.3.92
```
This error printing occured the I run the command "iw dev mesh0 mesh join my_mesh". Then, I can't ping the other mesh client. Note: All mesh interfaces start by default in channel 1. I've also tried changing the channel to 11 and 44.

This is the kernel log:
```
root@analog:~/openwifi# dmesg
Marvell 88E1510 e000b000.ethernet-ffffffff:00: attached PHY driver [Marvell 88E1510] (mii_bus:phy_addr=e000b000.ethernet-ffffffff:00, irq=POLL)
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
usbcore: registered new interface driver usb-storage
usbcore: registered new interface driver usbserial
usbcore: registered new interface driver usbserial_generic
usbserial: USB Serial support registered for generic
usbcore: registered new interface driver ftdi_sio
usbserial: USB Serial support registered for FTDI USB Serial Device
chipidea-usb2 e0002000.usb: e0002000.usb supply vbus not found, using dummy regulator
ULPI transceiver vendor/product ID 0x0424/0x0007
Found SMSC USB3320 ULPI transceiver.
ULPI integrity check: passed.
ci_hdrc ci_hdrc.0: EHCI Host Controller
ci_hdrc ci_hdrc.0: new USB bus registered, assigned bus number 1
ci_hdrc ci_hdrc.0: USB 2.0 started, EHCI 1.00
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 1 port detected
i2c /dev entries driver
at24 0-0050: 4096 byte 24c32 EEPROM, writable, 1 bytes/write
IR NEC protocol handler initialized
IR RC5(x/sz) protocol handler initialized
IR RC6 protocol handler initialized
IR JVC protocol handler initialized
IR Sony protocol handler initialized
IR SANYO protocol handler initialized
IR Sharp protocol handler initialized
IR MCE Keyboard/mouse protocol handler initialized
IR XMP protocol handler initialized
usbcore: registered new interface driver uvcvideo
USB Video Class driver (1.1.1)
gspca_main: v2.14.0 registered
cdns-wdt f8005000.watchdog: Xilinx Watchdog Timer at f0951000 with timeout 10s
Xilinx Zynq CpuIdle Driver started
sdhci: Secure Digital Host Controller Interface driver
sdhci: Copyright(c) Pierre Ossman
sdhci-pltfm: SDHCI platform and OF driver helper
mmc0: SDHCI controller on e0100000.mmc [e0100000.mmc] using ADMA
ledtrig-cpu: registered to indicate activity on CPUs
hidraw: raw HID events driver (C) Jiri Kosina
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
ad7291: probe of 0-002f failed with error -5
fpga_manager fpga0: Xilinx Zynq FPGA Manager registered
Netfilter messages via NETLINK v0.30.
mmc0: new high speed SDHC card at address b368
[... 312 lines cut ...]
IPv6: ADDRCONF(NETDEV_CHANGE): mesh: link becomes ready
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 0 M_BCN 1 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 0 BCN_PRP 0 PCP_FL 0 FCS_FL 0 ALL_MUT 1
mesh: TSF : max clockdrift=0; too small to adjust
sdr,sdr openwifi_prepare_multicast
sdr,sdr openwifi_configure_filter MON 0 M_BCN 1 BST0 1 BST1 1 UST 1 PB_RQ 0 PS_PL 1 O_BSS 1 CTL 0 BCN_PRP 0 PCP_FL 0 FCS_FL 0 ALL_MUT 1
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
mesh: TSF : max clockdrift=0; too small to adjust
```

twpedersen, 2022-08-28:
can you please share the output of:

`iw mesh0 station dump`

and

`ip a show mesh0`

Thanks.

yohanes-erwin, 2022-09-07:
This is the output of that commands:
`root@analog:~/openwifi# iw mesh0 station dump
root@analog:~/openwifi# ip a show mesh0
4: mesh0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 66:55:44:33:22:7a brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.92/24 brd 192.168.3.255 scope global mesh0
       valid_lft forever preferred_lft forever
    inet6 fe80::6455:44ff:fe33:227a/64 scope link
       valid_lft forever preferred_lft forever
root@analog:~/openwifi#`
The `iw mesh0 station dump` shows nothing.

JiaoXianjun, 2023-01-28:
Now it might support mesh (or you can enable it easily maybe), becaue we have upgraded to the Raspberry PI OS:

https://github.com/open-sdr/openwifi/releases/tag/v1.4.0

zhangmazi123321, 2023-02-23:
> Now it might support mesh (or you can enable it easily maybe), becaue we have upgraded to the Raspberry PI OS:
> 
> https://github.com/open-sdr/openwifi/releases/tag/v1.4.0
After updating the version, I still have the same problem as him. The configuration and errors are the same as his.

zhangmazi123321, 2023-02-23:
> This is the output of that commands:
>
> ```
> root@analog:~/openwifi# iw mesh0 station dump
> root@analog:~/openwifi# ip a show mesh0
> 4: mesh0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP [...]

Is your problem solved?

peking428, 2023-03-10:
thanks JiaoXianjun.   openwifi is very strong.  i finished this work that runing batman base on AP-sta/mesh/ad-hoc with this three mode.  debug this work, need some must step and ability of resolve problem. so. v :  peking428

xxw1617, 2023-09-14:
> Is your problem solved?
i have the same problem as you. Have you solved the problem?