open-sdr/openwifi #187: cannot find  the "openwifi" SSID

wwy9931154, 2022-07-01:
Hello, I run your image directly on zcu102 and the terminal print this message:Using interface sdr0 with hwaddr 66:55:44:33:22:72 and ssid "openwifi""sdr0: AP-ENABLED" so I thought the previous instruction has successfully run. However, I cannot find the "openwifi" SSID on my device. I tried some phone and I am sure that they support 5GHz. I used dmesg and read the message, I found some error message in it :
①"[    4.261217] ad7291: probe of 15-002f failed with error -5"
②"[    4.261217] ad7291: probe of 15-002f failed with error -5".
The following is all the information that the terminal feeds to me, which contains the information I obtained by using DMESG. What should I do to get the correct result? Really appreciate your help!

[logs from openwifi.docx](https://github.com/open-sdr/openwifi/files/9027722/logs.from.openwifi.docx)

wwy9931154, 2022-07-01:
I edited the log file that I reserved to make it easier to read.Now, the sentence I enter the instruction is marked as blue. The comments I manually entered in the log file are gray. The error information I found is red. 
[logs from openwifi.docx](https://github.com/open-sdr/openwifi/files/9027804/logs.from.openwifi.docx)

JiaoXianjun, 2022-07-06:
Can you search "tx_itrpt" in the issues to see how does it grow/no-grow in your case?

wwy9931154, 2022-07-07:
Thank you for your advice, I will try it immediately.

closed 2022-07-27 by JiaoXianjun