open-sdr/openwifi #191: Recompile sdr.ko run wgd.sh has problems

8663, 2022-07-21:
Getting this error when run wgd.sh:
Run wgd.sh appears when loads the recompiled driver ssdr.ko has a problem

root@analog:~/openwifi# ./wgd.sh

[image]

JiaoXianjun, 2022-07-21:
You need to describe step by step exact operations you have done from the beginning (also the output of the command you run). From downloading of software, our repository, SD card image, etc., to how do you compile the sdr.ko/kernel, how to put it onboard, etc.
You also need to describe the exact version of your OS, Vivado, etc.

8663, 2022-07-21:
The operating system we used was ubuntu16.04,Vivado is 2017.4
The whole process:
git init
git pull https://github.com/open-sdr/openwifi.git
programming environment setup：
export XILINX_DIR=/opt/Xilinx
source /opt/pkg/petalinux/settings.sh
source /opt/Xilinx/Vivado/2017.4/settings64.sh
[image]

[image]

JiaoXianjun, 2022-07-21:
Sorry that we do not have that Vivado version, and hard to reproduce the issue in the Vivado 2018.3 we suggested.

I suggest you use Ubuntu18 or 20 LTS together with Vivado 2018.3.

closed 2022-07-27 by 8663

chenmo234, 2024-06-04:
I also need to modify sdr.c and perform the same operations as you. Can I discuss this with you? Here is my contact information: c2385405729.